machinekit/QtQuickVcp #28: file upload fails on umlauts

strahlex, 2015-03-20:
When the uploaded file contains an umlaut the upload fails with an error.

sirop, 2016-02-28:
Maybe, this:
> ICU
> 
> From Qt 5.3 and onwards, configure does not link Qt Core against ICU libraries anymore by default. This reduces the size of a self-contained application package considerably.
> 
> Letting Qt Core utilize the ICU libraries however has following advantages:
> 
> Behavior matches other platforms more closely.
> Extended set of text codecs (see QTextCodec).

https://doc-snapshots.qt.io/qt5-5.6/windows-requirements.html

sirop, 2016-02-28:
Then this https://sourceforge.net/projects/qt64ng/files/qt/x86-64/5.5.0/msvc2013/
should be a remedy on Windows as this build inludes ICU in contrast to the official build 
from http://www.qt.io/download-open-source/ .

sirop, 2016-02-29:
@yishinli 
Could you please confirm that this bug is also valid for Chinese hieroglyphs.
If so, then we can be sure that it is about missing ICU libs in the officail Qt build.

strahlex, 2016-02-29:
At least I have not heard about it. Can you please test it ans send me the results. It's not unlikely that it is just a problem in QtQuickVcp and not Qt.

sirop, 2016-02-29:
I guess I have to follow https://github.com/strahlex/QtQuickVcp/wiki/Testing-mkwrapper 
to test it?

strahlex, 2016-02-29:
Yes, or use https://github.com/strahlex/mkwrapper-sim

sirop, 2016-03-01:
Ok, I tried the easy way:
```
slave@debian-slave:~/machinekit-sirop$ git clone https://github.com/thecooltool/machinekit-configs.git cooltool
slave@debian-slave:~/machinekit-sirop$ mklauncher cooltool/
```
The I get this on the MachinekitClient:  

[image]   

So what shall I do to provoke that error?
Rename UniStep* to something like Ünißtep*?

strahlex, 2016-03-01:
Upload a gcode file named äüß.ngc and see what happens

sirop, 2016-03-02:
Sorry for the delay, as I do it for the first time.

I did:
```
mkdir repos
git clone https://github.com/strahlex/Cetus.git
```
From another terminal over ssh:
` mklauncher mkwrapper-sim/` .

Then I copied the contents of https://github.com/thecooltool/machinekit-configs/blob/master/UNI-PRINT-3D/subroutines/g22.ngc  to `öäü.ngc` and `öab.ngc`. Tried to open these files over Cetus:
[image]

The third error in the pic is about `nöc_ballet_original.jpg`.  

On the remote machine with Machinekit installed:
```
cml@debian-cml2:~/repos$ ls -ls ../nc_files/*.ngc
4 -rw-r--r-- 1 cml cml 1593 Mar  2 04:54 ../nc_files/�ab.ngc
4 -rw-r--r-- 1 cml cml 1593 Mar  2 04:49 ../nc_files/���.ngc
0 -rw-r--r-- 1 cml cml    0 Mar  2 04:57 ../nc_files/nöc_ballet_original.ngc
```
It is strange that  *ö* was transmitted properly in case of `nöc_ballet_original.jpg` .

Maybe, we should not forget either that protobuf  - at least as used with machinekit - supports
only 7 bit ASCII.

sirop, 2016-03-06:
Just confirmed the test results with your http://buildbot.roessler.systems/files/machinekit-client/machinekit-client_v0.9.6-3-g75cd-126_windows_x86.zip build which  - as I understand - was built
with official Qt and therefore without ICU libs. Got the same errros with the three files as above.

So at least the main reason is not ICU within Qt.

yishinli, 2016-03-07:
> @yishinli
> Could you please confirm that this bug is also valid for Chinese hieroglyphs.

The Machinekit-client can not display Chinese words correctly. Please take a look of the following picture:
[image]
The Chinese file name should be 2016-03-07_週期測試.ngc

I've set LANG variables before starting machinekit-client.
```
export LANG=zh_TW.UTF-8
export LANGUAGE=zh_TW.UTF-8
```

Regards,

machinekoder, 2017-09-07:
@yishinli I think that's a different issue. I think Qt needs to be configured to support Chinese characters so it might be good to create a separate issue.