machinekit/QtQuickVcp #30: Possible explanation for nonfunctional ZoomIn/ZoomOut

RunningLight, 2015-04-23:
I apologize for not yet having a working development environment in which to test this hypothesis, but in code-walking I note that `./src/pathview/ZoomInAction.qml` and its mates `ZoomOriginalAction.qml` and `ZoomOutAction.qml` all set `view.cameraZoom` but `./src/pathview/PathView3D.qml` uses `pathview.cameraZoom`.

Might this be the reason the respective Zoom buttons in Cetus are not functioning?

strahlex, 2015-04-23:
Good direction. The problem was actually that I forgot to update to rename the zoom property in Cetus. Fix is here: https://github.com/strahlex/Cetus/commit/ad55915e876f3f3b8be44754ebaf7d94fee61311

closed 2015-04-23 by strahlex

RunningLight, 2015-04-23:
Hmmm. I'm still trying to find my sea legs with Qt and qml, obviously. Thanks for the quick fix and the huge amount of work you've put into QtQuickVcp.