machinekit/QtQuickVcp #34: Android App on Android 5.0 - service discovery not working

koppi, 2015-04-25:
[image]

koppi, 2015-04-25:
Bonjour Browser app works:

[image]

strahlex, 2015-04-27:
Detecting networks happens here in the code https://github.com/strahlex/QtQuickVcp/blob/master/src/service/qservicediscovery.cpp#L1220 I will need to test this with an Android 5.0 device.

strahlex, 2015-05-03:
Tested with Cyanogenmod 12 on a Nexus 10 and it worked. So the problem seems depend on the hardware as well. 

Today I found that if the network supports IPv6 announcement happen in the IPv6 range. This may result in the instance not showing up. However, in my case Bonjour browser also showed no services so this might not be the case.

What network types (WiFi, 3G, ..) do you have enabled?

koppi, 2015-05-03:
```
$ adb shell netcfg
wlan0    UP                                192.168.0.10/24  0x00001043 40:b0:fa:69:f9:84
rmnet_usb0 DOWN                                   0.0.0.0/0   0x00000000 00:00:00:00:00:00
rmnet_usb2 DOWN                                   0.0.0.0/0   0x00000000 00:00:00:00:00:00
rmnet_usb1 DOWN                                   0.0.0.0/0   0x00000000 00:00:00:00:00:00
rmnet_usb3 DOWN                                   0.0.0.0/0   0x00000000 00:00:00:00:00:00
lo       UP                                   127.0.0.1/8   0x00000049 00:00:00:00:00:00
sit0     DOWN                                   0.0.0.0/0   0x00000080 00:00:00:00:00:00
p2p0     UP                                     0.0.0.0/0   0x00001003 42:b0:fa:69:f9:84
dummy0   DOWN                                   0.0.0.0/0   0x00000082 6a:6c:5f:4f:42:9c
rmnet0   DOWN                                   0.0.0.0/0   0x00001002 f2:d5:7d:0b:1e:20
rmnet1   DOWN                                   0.0.0.0/0   0x00001002 c6:a6:a1:fc:27:fe
rmnet3   DOWN                                   0.0.0.0/0   0x00001002 3e:1d:1b:bb:ad:32
rmnet2   DOWN                                   0.0.0.0/0   0x00001002 66:d7:d4:53:c8:66
rmnet4   DOWN                                   0.0.0.0/0   0x00001002 4a:e8:87:e1:20:e7
rmnet6   DOWN                                   0.0.0.0/0   0x00001002 42:65:09:ee:58:47
rmnet5   DOWN                                   0.0.0.0/0   0x00001002 ca:33:c5:df:d6:38
rmnet7   DOWN                                   0.0.0.0/0   0x00001002 1e:af:f2:3c:b3:9c
rev_rmnet3 DOWN                                   0.0.0.0/0   0x00001002 a6:92:23:d4:35:3e
rev_rmnet2 DOWN                                   0.0.0.0/0   0x00001002 f2:ab:c1:8f:c6:f9
rev_rmnet4 DOWN                                   0.0.0.0/0   0x00001002 fa:02:2c:39:48:af
rev_rmnet6 DOWN                                   0.0.0.0/0   0x00001002 a2:f4:0f:f7:52:7d
rev_rmnet5 DOWN                                   0.0.0.0/0   0x00001002 7a:0c:28:74:36:8f
rev_rmnet7 DOWN                                   0.0.0.0/0   0x00001002 4e:d9:9d:db:6b:6a
rev_rmnet8 DOWN                                   0.0.0.0/0   0x00001002 ca:aa:13:f2:08:ca
rev_rmnet0 DOWN                                   0.0.0.0/0   0x00001002 06:69:0f:e8:04:81
rev_rmnet1 DOWN                                   0.0.0.0/0   0x00001002 42:56:2c:e2:c8:03
rmnet_smux0 DOWN                                   0.0.0.0/0   0x00001002 0a:0a:0e:5c:85:bf
```

and

```
$ adb shell ip addr show wlan0
26: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP qlen 100
    link/ether 40:b0:fa:69:f9:84 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.10/24 brd 192.168.0.255 scope global wlan0
    inet6 fd00:c47:3d97:1b32:706b:6768:7d19:79f2/64 scope global temporary dynamic 
       valid_lft 535426sec preferred_lft 77317sec
    inet6 fd00:c47:3d97:1b32:42b0:faff:fe69:f984/64 scope global dynamic 
       valid_lft 535426sec preferred_lft 401569sec
    inet6 2a02:810d:20bf:e30c:706b:6768:7d19:79f2/64 scope global temporary dynamic 
       valid_lft 5398sec preferred_lft 2698sec
    inet6 2a02:810d:20bf:e30c:42b0:faff:fe69:f984/64 scope global dynamic 
       valid_lft 5398sec preferred_lft 2698sec
    inet6 fe80::42b0:faff:fe69:f984/64 scope link 
       valid_lft forever preferred_lft forever
```

koppi, 2015-05-04:
Serivce discovery still does not work ipv6 patches from @mhaberler applied to machinekit. So there might remain the issue with the Qt / Android 5.0 combination.

strahlex, 2015-05-04:
I think I know what the problem is. On your device multicast seems to be disabled or not working. ServiceDiscovery works at the moment in a way that it displays the network not found error when multicast does not work, so you can not switch to unicast. The problem has nothing todo with IPv6.

To fix this problem might take some time.

strahlex, 2015-07-08:
Can please try again with the latest and greatest Qt5.5 based version: http://buildbot.roessler.systems/files/machinekit-client/

koppi, 2015-07-08:
On the ML I've read that something changed in the way, mk.py needs to be written. Is there any "Hello World" config repo to get a quick start for testing?

koppi, 2015-07-08:
@strahlex is the info at https://github.com/strahlex/QtQuickVcp/wiki/Testing-mkwrapper still valid?

strahlex, 2015-07-08:
Yes, I have updated the wiki. Take a look at https://github.com/strahlex/MendelMax-CRAMPS and https://github.com/thecooltool/Uni-print-3D for the Python based 3D printer configs.

strahlex, 2015-07-08:
@koppi That is the additional service that is now necessary to connect https://github.com/strahlex/QtQuickVcp/wiki/Using-mklauncher

koppi, 2015-07-08:
@strahlex multicast service discovery works now with the latest version:
http://buildbot.roessler.systems/files/machinekit-client/machinekit-client_v0.9.2-6_android_armv7.apk
[image]

koppi, 2015-07-08:
I just gave a try to https://github.com/strahlex/anddemo
There are no services displayed, see logs and screenshot:
```bash
$ mklauncher -d .
announcing mklauncher
launcher {
  index: 0
  name: "AND Demo"
  description: "Demonstrating AND in HAL"
  image {
    name: "andgate.png"
    encoding: CLEARTEXT
    blob: "\211PNG\r\n\032\n\000...
\256B`\202"
  }
  info {
    type: "Demo"
    manufacturer: "Alex"
    model: ""
    variant: "default"
  }
  running: false
  terminating: false
  command: "python run.py -c"
  shell: false
  returncode: 0
  workdir: "."
}
launcher {
  index: 1
  name: "AND Demo with HAL Scope and HAL Meter"
  description: "Demonstrating AND in HAL"
  image {
    name: "andgate.png"
000IEND\256B`\202"
  }
  info {
    type: "Demo"
    manufacturer: "Alex"
    model: ""
    variant: "with HAL Scope and HAL Meter"
  }
  running: false
  terminating: false
  command: "python run.py -c -s -m"
  shell: false
  returncode: 0
  workdir: "."
}

service: dsname = tcp://x220.local.:62440 port = 62440 txtrec = ['dsn=tcp://x220.local.:62440', 'uuid=a42c8c6b-4025-4f83-ba28-dad21114744a', 'instance=6c455aba-25c8-11e5-9d57-9b7ad11dc39b', 'service=launcher'] name = Machinekit Launcher on x220.local.
service: dsname = tcp://x220.local.:53774 port = 53774 txtrec = ['dsn=tcp://x220.local.:53774', 'uuid=a42c8c6b-4025-4f83-ba28-dad21114744a', 'instance=6c455abb-25c8-11e5-9d57-9b7ad11dc39b', 'service=launchercmd'] name = Launchercmd service on x220.local. pid 21684
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
sending launcher message
^Cstopping threads
threads stopped
```
adb logcat:
```
I/ActivityManager(  557): START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10200000 pkg=io.machinekit.appdiscover cmp=io.machinekit.appdiscover/org.qtproject.qt5.android.bindings.QtActivity bnds=[216,785][376,985] (has extras)} from uid 10016 on display 0
D/audio_hw_primary(  197): select_devices: out_snd_device(2: speaker) in_snd_device(0: none)
D/ACDB-LOADER(  197): ACDB -> send_afe_cal
D/audio_hw_primary(  197): enable_snd_device: snd_device(2: speaker)
D/audio_hw_primary(  197): enable_audio_route: apply and update mixer path: low-latency-playback
W/libmachinekit-client.so( 8766): (null):0 ((null)): Can't find surface 8
W/libmachinekit-client.so( 8766): (null):0 ((null)): Can't find surface 8
I/Keyboard.Facilitator(  947): onFinishInput()
V/WindowManager(  557): Adding window Window{146457f9 u0 SurfaceView} at 8 of 16 (before Window{2587b094 u0 SurfaceView EXITING})
D/audio_hw_primary(  197): disable_audio_route: reset and update mixer path: low-latency-playback
D/audio_hw_primary(  197): disable_snd_device: snd_device(2: speaker)
```
Screenshot:
[image]

strahlex, 2015-07-09:
@koppi You are already connected. Depending on the speed of your network connection it can take some time until the launchers are loaded (there should be a loading indicator...). Can you please try to comment out the images in the `launcher.ini` file

koppi, 2015-07-09:
I commented out the images in ```launcher.ini``` kept the loading dialog open for about 10 minutes, the app still shows no launch items. Same result with the latest Linux version of machinekit-client:
[image]

strahlex, 2015-07-09:
Then there must be something wrong with your host. Any firewalls activated?

koppi, 2015-07-09:
```bash
koppi@x220:~$ sudo iptables -L
Chain INPUT (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     udp  --  anywhere             anywhere             udp dpt:domain
ACCEPT     tcp  --  anywhere             anywhere             tcp dpt:domain
ACCEPT     udp  --  anywhere             anywhere             udp dpt:bootps
ACCEPT     tcp  --  anywhere             anywhere             tcp dpt:bootps

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     all  --  anywhere             192.168.122.0/24     ctstate RELATED,ESTABLISHED
ACCEPT     all  --  192.168.122.0/24     anywhere            
ACCEPT     all  --  anywhere             anywhere            
REJECT     all  --  anywhere             anywhere             reject-with icmp-port-unreachable
REJECT     all  --  anywhere             anywhere             reject-with icmp-port-unreachable

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         
ACCEPT     udp  --  anywhere             anywhere             udp dpt:bootpc
```

strahlex, 2015-07-09:
I wonder at which stage this fails. Can you please try a `avahi-browse -a` on your Ubuntu machine. The package you will need is `avahi-utils`.

koppi, 2015-07-09:
```bash
koppi@x220:~$ avahi-browse -a 
+ virbr0 IPv4 Launchercmd service on x220.local. pid 19664  _machinekit._tcp     local
+ virbr0 IPv4 Machinekit Launcher on x220.local.            _machinekit._tcp     local
+  wlan0 IPv4 Launchercmd service on x220.local. pid 19664  _machinekit._tcp     local
+  wlan0 IPv4 Machinekit Launcher on x220.local.            _machinekit._tcp     local
+ virbr0 IPv4 x220-2 [52:54:00:e1:d7:9f]                    Workstation          local
+  wlan0 IPv6 x220-2 [10:0b:a9:bc:e1:64]                    Workstation          local
+  wlan0 IPv4 x220-2 [10:0b:a9:bc:e1:64]                    Workstation          local
^CGot SIGINT, quitting.
```

koppi, 2015-07-09:
```bash
koppi@x220:~$ ping x220.local
PING x220.local (192.168.0.2) 56(84) bytes of data.
64 bytes from x220.local (192.168.0.2): icmp_seq=1 ttl=64 time=0.036 ms
^C
--- x220.local ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 0.036/0.036/0.036/0.000 ms
```

strahlex, 2015-07-09:
Ah, here we have the problem. The mDNS hostname is colliding with another one your network.

@mhaberler This is what you have solved with the last zeroconf update for HAL, right?

mhaberler, 2015-07-09:
in theory... but I could have botched it. I guess the renamed hostname should go into the announcement, right?

how do I force this, have a second host with the same hostname and avahi-demon on the same wire I guess?

strahlex, 2015-07-09:
I have not adopted this behavior to the Python based announcements yet. So this could clearly be an indicator this problem is happening.

@koppi Can you please start the run.py manually and do another `avahi-browse -a` so we can see what haltalk announces.

koppi, 2015-07-09:
I just ran the ```mklauncher -d .``` on my x200 (Debian/Jessie), it works:
[image]

mhaberler, 2015-07-09:
well first the log should show the actual hostname during startup: https://github.com/machinekit/machinekit/blob/master/src/machinetalk/lib/ll_zeroconf_register.cc#L43

so this should show up in the log as x220-2 or somesuch

this is the location where the detected (renamed) hostname is interpolated into the zmq URI: https://github.com/machinekit/machinekit/blob/master/src/machinetalk/lib/ll_zeroconf_register.cc#L56-L60

heelp! dont shoot at the piano player ;)

mhaberler, 2015-07-09:
@koppi - we need the actual URI's announced, I use avahi-discover for that

koppi, 2015-07-09:
@mhaberler see
```bash
koppi@x220:~$ avahi-discover 
Browsing domain 'local' on -1.-1 ...
Browsing for services of type '_machinekit._tcp' in domain 'local' on 5.0 ...
Browsing for services of type '_workstation._tcp' in domain 'local' on 5.0 ...
Browsing for services of type '_machinekit._tcp' in domain 'local' on 4.1 ...
Browsing for services of type '_workstation._tcp' in domain 'local' on 4.1 ...
Browsing for services of type '_machinekit._tcp' in domain 'local' on 4.0 ...
Browsing for services of type '_workstation._tcp' in domain 'local' on 4.0 ...
Found service 'Launchercmd service on x220.local. pid 22501' of type '_machinekit._tcp' in domain 'local' on 5.0.
Found service 'Machinekit Launcher on x220.local.' of type '_machinekit._tcp' in domain 'local' on 5.0.
Found service 'x220-2 [52:54:00:e1:d7:9f]' of type '_workstation._tcp' in domain 'local' on 5.0.
Found service 'Log service on x200.local pid 3551' of type '_machinekit._tcp' in domain 'local' on 4.1.
Found service 'x200 [00:21:6a:b8:23:ca]' of type '_workstation._tcp' in domain 'local' on 4.1.
Found service 'x220-2 [10:0b:a9:bc:e1:64]' of type '_workstation._tcp' in domain 'local' on 4.1.
Found service 'Launchercmd service on x220.local. pid 22501' of type '_machinekit._tcp' in domain 'local' on 4.0.
Found service 'Machinekit Launcher on x220.local.' of type '_machinekit._tcp' in domain 'local' on 4.0.
Found service 'AND-Demo on x200.local.' of type '_machinekit._tcp' in domain 'local' on 4.0.
Found service 'x200 [00:21:6a:b8:23:ca]' of type '_workstation._tcp' in domain 'local' on 4.0.
Found service 'x220-2 [10:0b:a9:bc:e1:64]' of type '_workstation._tcp' in domain 'local' on 4.0.
```
[image]

strahlex, 2015-07-09:
@koppi It's correct on your x220 host. Can you please try to run the anddemo example directly from the command line on the conflicting x220-2 host.

mhaberler, 2015-07-09:
click on one of the _machinekit._tcp lines, then you see the full announcenemt:

[image]

the hostname in the TXT dsn record counts

koppi, 2015-07-09:
I was doing a "ping x220.local", it said
```
ping: unknown host x220.local
```
I was assuming, that an entry in ```/etc/hosts``` is required for ```x220.local```, this is where the ```x220-2``` probably came from. I just removed the x220.entry entry from ```/etc/hosts```.

mhaberler, 2015-07-09:
hm well avahi renames hosts on the fly if the are collisions

koppi, 2015-07-09:
[image]

koppi, 2015-07-09:
@strahlex on my Ubuntu box I ran into another unrelated issue:
```bash
koppi@x220:~/anddemo$ python run.py 
Traceback (most recent call last):
  File "run.py", line 9, in <module>
    from machinekit import rtapi
  File "hal/cython/machinekit/rtapi.pyx", line 139, in init machinekit.rtapi (hal/cython/machinekit/rtapi.c:7433)
ImportError: /usr/lib/python2.7/dist-packages/machinekit/hal.so: undefined symbol: hal_shmem_base
```

mhaberler, 2015-07-09:
@koppi uh I smell another Ubuntu "improvement" on linker semantics. Do you now understand why we dumped this caricature of userfriendly.com ;-?

mah@nwheezy:~/machinekit-icomp$ nm lib/python/machinekit/hal.so |grep shmem_
         U hal_shmem_base

mah@nwheezy:~/machinekit-icomp$ ldd lib/python/machinekit/hal.so 
	linux-gate.so.1 (0xb76e2000)
	liblinuxcnchal.so.0 => /home/mah/machinekit-icomp/lib/liblinuxcnchal.so.0 (0xb763c000)

------^^^^ hal_shmem_base must be exported here:

	libmtalk.so.0 => /home/mah/machinekit-icomp/lib/libmtalk.so.0 (0xb761d000)
	libgcc_s.so.1 => /lib/i386-linux-gnu/libgcc_s.so.1 (0xb75dd000)
	libc.so.6 => /lib/i386-linux-gnu/i686/cmov/libc.so.6 (0xb7433000)
	liblinuxcnc-pb2++.so.0 => /home/mah/machinekit-icomp/lib/liblinuxcnc-pb2++.so.0 (0xb7229000)
	libstdc++.so.6 => /usr/lib/i386-linux-gnu/libstdc++.so.6 (0xb7137000)
	libdl.so.2 => /lib/i386-linux-gnu/i686/cmov/libdl.so.2 (0xb7132000)
	librt.so.1 => /lib/i386-linux-gnu/i686/cmov/librt.so.1 (0xb7128000)
	libprotobuf.so.9 => /usr/lib/i386-linux-gnu/libprotobuf.so.9 (0xb6ffd000)
	libpthread.so.0 => /lib/i386-linux-gnu/i686/cmov/libpthread.so.0 (0xb6fe1000)
	libczmq.so.3 => /usr/local/lib/libczmq.so.3 (0xb6f2b000)
	libzmq.so.3 => /usr/lib/i386-linux-gnu/libzmq.so.3 (0xb6ec6000)
	liblinuxcncini.so.0 => /home/mah/machinekit-icomp/lib/liblinuxcncini.so.0 (0xb6ec1000)
	libavahi-common.so.3 => /usr/lib/i386-linux-gnu/libavahi-common.so.3 (0xb6eb2000)
	libavahi-client.so.3 => /usr/lib/i386-linux-gnu/libavahi-client.so.3 (0xb6e9e000)
	libuuid.so.1 => /lib/i386-linux-gnu/libuuid.so.1 (0xb6e98000)
	libjansson.so.4 => /usr/lib/i386-linux-gnu/libjansson.so.4 (0xb6e89000)
	libm.so.6 => /lib/i386-linux-gnu/i686/cmov/libm.so.6 (0xb6e42000)
	/lib/ld-linux.so.2 (0xb76e3000)
	libz.so.1 => /lib/i386-linux-gnu/libz.so.1 (0xb6e25000)
	libpgm-5.1.so.0 => /usr/lib/libpgm-5.1.so.0 (0xb6dcf000)
	libsodium.so.13 => /usr/lib/i386-linux-gnu/libsodium.so.13 (0xb6d50000)
	libdbus-1.so.3 => /lib/i386-linux-gnu/libdbus-1.so.3 (0xb6cf8000)

and yes, it is:

mah@nwheezy:~/machinekit-icomp$ nm /home/mah/machinekit-icomp/lib/liblinuxcnchal.so.0|grep shmem
00025940 B hal_shmem_base

koppi, 2015-07-09:
```bash
$ ldd /usr/lib/python2.7/dist-packages/machinekit/hal.so
	linux-vdso.so.1 =>  (0x00007ffd091d1000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fac76819000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fac7644f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fac76cc8000)
```

koppi, 2015-07-09:
I'm closing this issue, as I'm now able to connect to my Mk instance on x200 (Debian / Jessie).

closed 2015-07-09 by koppi

mhaberler, 2015-07-09:
@koppi I suggest filing a separate issue for the Ubuntu problem, this does not belong here and closing it does not make sense

also, could you make a stab and try the steps I outlined above?

strahlex, 2015-07-09:
See https://github.com/machinekit/machinekit/pull/707

koppi, 2015-07-09:
@strahlex when I run with your latest two patches I see
```bash
koppi@x220:~/anddemo$ mklauncher -d .
Remote communication is deactivated, configserver will use the loopback interfaces
set REMOTE in /etc/linuxcnc/machinekit.ini to 1 to enable remote communication
announcing mklauncher
launcher {
  index: 0
  name: "AND Demo"
  description: "Demonstrating AND in HAL"
  info {
    type: "Demo"
    manufacturer: "Alex"
    model: ""
    variant: "default"
  }
  running: false
  terminating: false
  command: "python run.py -c"
  shell: false
  returncode: 0
  workdir: "."
}
launcher {
  index: 1
  name: "AND Demo with HAL Scope and HAL Meter"
  description: "Demonstrating AND in HAL"
  info {
    type: "Demo"
    manufacturer: "Alex"
    model: ""
    variant: "with HAL Scope and HAL Meter"
  }
  running: false
  terminating: false
  command: "python run.py -c -s -m"
  shell: false
  returncode: 0
  workdir: "."
}

service: dsname = tcp://127.0.0.1:63576 port = 63576 txtrec = ['dsn=tcp://127.0.0.1:63576', 'uuid=a42c8c6b-4025-4f83-ba28-dad21114744a', 'instance=198e399e-265a-11e5-9d57-9b7ad11dc39b', 'service=launcher'] name = Machinekit Launcher on %(fqdn)s
service: dsname = tcp://127.0.0.1:56638 port = 56638 txtrec = ['dsn=tcp://127.0.0.1:56638', 'uuid=a42c8c6b-4025-4f83-ba28-dad21114744a', 'instance=198e399f-265a-11e5-9d57-9b7ad11dc39b', 'service=launchercmd'] name = Launchercmd service on %(fqdn)s pid 22705
sending launcher message
sending launcher message
...
```

Is the line ``` 'service=launchercmd'] name = Launchercmd service on %(fqdn)s pid 22705``` correct?

koppi, 2015-07-09:
I can confirm, launcher discovery works now:
[image]
But somehow the launcher icons ware missing:
[image]

strahlex, 2015-07-09:
@koppi  The line is correct. fqdn is replaced during the service announcement with the correct hostname.