machinekit/QtQuickVcp #37: PathView3D drawing error for G3

f1oat, 2015-07-13:
Here is a simple test sequence showing the issue on G3.

**G21 G90 G64 G40
G17
F100
G0 X0 Y0
G91
M3 S1000
G1 X0 Y0
G2 X10 Y10 I10
G3 X10 Y-10 I10
M5
M2**

Result with Axis: 
[image]
Result with PathView3D: 
[image]

Pathview3D is drawing "G3 X10 **Y-10** I10" as if it was "G3 X10 **Y+10** I10"

strahlex, 2015-07-14:
The relevant code is here: https://github.com/strahlex/QtQuickVcp/blob/master/src/pathview/qglpathitem.cpp#L482

strahlex, 2015-07-22:
fixed by https://github.com/strahlex/QtQuickVcp/commit/88a7a019507bb92aa3604d83531a32c253af0618

strahlex, 2016-07-20:
can we close this?

f1oat, 2016-07-24:
I have to check with up to date repo.
As far as I remember, there were a graphical glitch on another portion 
of the drawing.

closed 2017-11-01 by machinekoder