machinekit/QtQuickVcp #39: Does the qtquickvcp_install.sh have to install Qt SDK?

juodumas, 2015-10-12:
[QtQuick-Virtual-Control-Panel](https://github.com/strahlex/QtQuickVcp/wiki/QtQuick-Virtual-Control-Panel) links to [qtquickvcp_install.sh](https://gist.github.com/strahlex/e4b9137710e23723ab7a), which installs the whole Qt5 SDK. Is there a reason for this? The script below is enough to compile and install QtQuickVCP and MachinekitClient to Debian wheezy and later (though I didn't try connecting the MachinekitClient to anything). Or does QtQuickVCP require the latest Qt release (wheezy-backports and jessie has Qt 5.3.2)?

```sh
# If running Debian wheezy, make sure you have the wheezy-backports repo,
# i.e. this line must be present in /etc/apt/sources.list:
# deb http://httpredir.debian.org/debian wheezy-backports main

# Add machinekit's repo and keyring to apt
sudo sh -c 'echo deb http://deb.dovetail-automata.com wheezy main > /etc/apt/sources.list.d/machinekit.list'
sudo apt-get update
sudo apt-get install dovetail-automata-keyring
sudo apt-get update

# Install gcc, git, qt5, protobuf, zmq
sudo apt-get install -y build-essential git qt5-default qtdeclarative5-dev protobuf-compiler libzmq3-dev

# Compile and install QtQuickVcp
git clone https://github.com/strahlex/QtQuickVcp
cd QtQuickVcp
qmake
make
sudo make install
cd ..

# Install required QML modules for MachinekitClient
sudo apt-get install -y qml-module-qtquick2 qml-module-qtquick-controls qml-module-qt-labs-settings qml-module-qtquick-dialogs qml-module-qtquick-particles2 qml-module-qtquick-xmllistmodel
# Compile and install MachinekitClient
git clone https://github.com/strahlex/MachinekitClient
cd MachinekitClient
qmake
make
sudo make install
cd ..

# You might want to install qtcreator to use the QtQuickVcp components.
# Run this on Debian Jessie and later:
# sudo apt-get install -y qtcreator
# Or run this on Debian wheezy:
# sudo apt-get -t wheezy-backports install qtcreator
```

strahlex, 2015-10-12:
Qt 5.3.2 is enough to compile QtQuickVcp. However, Qt is evolving quickly and especially QtCreator has improved a lot over the last years. Since 3.5.0 the QtQuick designer finally works. You do not need the Machinekit SDK to work with QtQuickVcp.

The install script is not 100% up to date. I am working on a vagrant based solution for the QtQuickVcp SDK https://github.com/machinekit/machinekit/issues/765 Depending on how urgent you need to use QtQuickVcp I encourage you to wait for the Novemeber meetup, since I will give a talk about that topic. I will release a vagrant configuration for testing this or next week.

strahlex, 2016-07-03:
Did you test the Vagrant solution? Can I close this issue?

juodumas, 2016-07-04:
Feel free to close this issue.

I haven't tested your Vagrantfile. The script in my initial comment
was enough to compile QtQuickVcp for me, but it has been a while..

closed 2016-07-05 by strahlex