machinekit/QtQuickVcp #40: Debian Jessie - Service discover not working

yishinli, 2015-12-26:
The Service Discover stops working after upgrading kernel from 3.16.7-ckt11-1+deb8u6 to 3.16.7-ckt20-1+deb8u1. 
* avahi-discover does find the Machinekit services. 
* affected device is wlan0
* with eht0, the Service Discover does work.
* switch back to old kernel works.
* seems related to [#34](https://github.com/strahlex/QtQuickVcp/issues/34)

Any advice on how to identify/solve this issue?

OS: Debian Jessie
```
uname -a: 
Linux interlaken 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt20-1+deb8u1 (2015-12-14) x86_64 GNU/Linux
```
[image]

strahlex, 2015-12-26:
Is `eth0` active at the same time? It is possible that JDNS queries only work on one interface.

yishinli, 2015-12-27:
I upgraded to 3.16.7-ckt20-1+deb8u1 kernel again, and do the following settings:
* eth0 is not active, the Ethernet port was not connected.
* vmnet* (the Ethernet device of VMWard) are turned off
* the only activated network device is wlan0

=> The service discover still doesn't work with wlan0.

Is there any options that I should enable to get meaningful log messages?

strahlex, 2015-12-27:
Yes, run MachinekitClient from the terminal.

dschbrs, 2016-01-04:
I can see this same problem after a system update in my computer  3.16.0-57-generic #77~14.04.1-Ubuntu SMP. Before the update Qtquickvcp worked like a charm.  The kernel was   3.16.0-30-generic #77~14.04.1-Ubuntu SMP. Luckily I did a partition cloning before the update. With the backup Qtquickvcp continue working OK. Running from a terminal no significant info:
qml: Starting to load the main application.
qml: Main application loaded.
qml: Finishing the splash screen progress bar.
qml: Hiding the splash screen.
qml: Deleting the splash screen.

Multicast or unicast not working.

strahlex, 2016-01-04:
Can you check the Qt version. Qt 5.5 and 5.5.1 have a bug in the networking library that prevents service discovery from working. The bug will be fixed with version 5.6 and 5.5.2.

dschbrs, 2016-01-04:
The server side is running Debian 3.8-1-xenomai.x86-686-pae Debian
3.8.13.-12   1wheezy 1da i686. The QT is 5.4 installed in July 2015. This
work flawless combined with Ubuntu before the kernel update and continue
working with my backup partition.

yishinli, 2016-01-10:
@strahlex 

> Can you check the Qt version. Qt 5.5 and 5.5.1 have a bug in the networking library that prevents service discovery from working. The bug will be fixed with version 5.6 and 5.5.2.

After upgrading to Qt5.6.0, the "Service discover issue" no longer appear for up-to-date kernerl(3.16.7-ckt20-1+deb8u2)

However, with Qt5.6.0, there will be "HelperWidgets" error for qmlplugindump.
```
qmlplugindump -v Machinekit.HalRemote 1.0 imports/
...
QQmlComponent: Component is not ready
...../imports/loaddependencies.qml:2:1: module "HelperWidgets" is not installed
```
My workaround is making a symbolic link for HelperWidgets from qtCreator to qml/ directory.
Since HelperWidgets is not the standard qml package, is it necessary for QtQuickVcp?

strahlex, 2016-01-10:
Looks like this bug is also present in earlier versions with newer kernel. Too bad the 5.6 release is still a month away.

strahlex, 2016-02-25:
Qt 5.6 is on its way. The RC looks fine.

strahlex, 2016-03-21:
Qt 5.6 has been released. @yishinli Can you confirm that this problem has been solved.

yishinli, 2016-03-24:
Building QtQuickVcp with Qt 5.6 solves this issue.

closed 2016-03-24 by yishinli