machinekit/QtQuickVcp #43: Merge MachinekitClient back to QtQuickVcp

strahlex, 2016-01-11:
Having Machinekit in its own repository seems to be confusing for many users. The benefits are trivial and insignificant and merging both repository may reduce complexity. https://github.com/strahlex/MachinekitClient

However, it may not be acceptable to have it as subproject of QtQuickVcp since this could affect deployment. The other case -> having QtQuickVcp as subproject of MachinekitClient might be a viable way. `make install` was usually the problem.

Another solution might be to create a master repository with everything as submodules like https://github.com/strahlex/machinetalk-sandbox

sirop, 2016-02-29:
> The other case -> having QtQuickVcp as subproject of MachinekitClient might be a viable way. make install was usually the problem.

This middle way is my favourite as one has then an application which - if compiled and executed - is
a minimal proof for all the prerequisite libs (protobuffer, zeromq) built and installed properly.

> Another solution might be to create a master repository with everything as submodules like https://github.com/strahlex/machinetalk-sandbox 

Do not know if it is too much work. If it were easy to choose which submodules are to built and which
are to be excluded from build process, then this proposal is also like "having QtQuickVcp as subproject of MachinekitClient" .

machinekoder, 2016-09-25:
I added MachinekitClient now back to QtQuickVcp. I hope this decreases confusion.

closed 2016-09-25 by machinekoder