machinekit/QtQuickVcp #44: Could not build QtQuickVcp

zhivko, 2016-01-11:
I am trying to build:
QtQuickVcp

Qmake version:
kz@klemenHp:~/git/QtQuickVcp$ qmake --version
QMake version 3.0
Using Qt version 5.4.1 in /usr/lib/i386-linux-gnu

I am on:
kz@klemenHp:~/git/QtQuickVcp$ uname -a
Linux klemenHp 3.19.0-42-generic #48-Ubuntu SMP Thu Dec 17 22:55:02 UTC 2015 i686 i686 i686 GNU/Linux

I get following:
```
kz@klemenHp:~/git/QtQuickVcp$ qmake QtQuickVcp.pro 
kz@klemenHp:~/git/QtQuickVcp$ make
cd 3rdparty/ && ( test -e Makefile || /usr/lib/i386-linux-gnu/qt5/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/3rdparty/3rdparty.pro -o Makefile ) && make -f Makefile 
make[1]: Entering directory '/mnt/Extra10Gb/git/QtQuickVcp/3rdparty'
cd machinetalk-protobuf-qt/ && ( test -e Makefile || /usr/lib/i386-linux-gnu/qt5/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/3rdparty/machinetalk-protobuf-qt/machinetalk-protobuf-qt.pro -o Makefile ) && make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/QtQuickVcp/3rdparty/machinetalk-protobuf-qt'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/3rdparty/machinetalk-protobuf-qt'
make[1]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/3rdparty'
cd src/ && ( test -e Makefile || /usr/lib/i386-linux-gnu/qt5/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/src.pro -o Makefile ) && make -f Makefile 
make[1]: Entering directory '/mnt/Extra10Gb/git/QtQuickVcp/src'
cd halremote/ && ( test -e Makefile || /usr/lib/i386-linux-gnu/qt5/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/halremote/halremote.pro -o Makefile ) && make -f Makefile 
Project ERROR: Unknown module(s) in QT: qml quick
Makefile:49: recipe for target 'sub-halremote-make_first-ordered' failed
make[1]: *** [sub-halremote-make_first-ordered] Error 3
make[1]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/src'
Makefile:86: recipe for target 'sub-src-make_first-ordered' failed
make: *** [sub-src-make_first-ordered] Error 2
```

sirop, 2016-01-11:
Could http://askubuntu.com/a/508524 be the answer?
> `sudo apt-get install qtdeclarative5-dev`

zhivko, 2016-01-14:
Yes that helped, but then I got:
```
Makefile install
make[2]: Entering directory '/mnt/Extra10Gb/git/QtQuickVcp/src/halremote'
/usr/lib/i386-linux-gnu/qt5/bin/qmlplugindump -nonrelocatable Machinekit.HalRemote 1.0 /mnt/Extra10Gb/git/QtQuickVcp/src/halremote/../../imports/ > /mnt/Extra10Gb/git/QtQuickVcp/src/halremote/../../imports/Machinekit/HalRemote/plugins.qmltypes
/bin/sh: 1: /usr/lib/i386-linux-gnu/qt5/bin/qmlplugindump: not found
Makefile:653: recipe for target 'install_dumppluginqmltypes' failed
make[2]: *** [install_dumppluginqmltypes] Error 127
make[2]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/src/halremote'
Makefile:73: recipe for target 'sub-halremote-install_subtargets-ordered' failed
make[1]: *** [sub-halremote-install_subtargets-ordered] Error 2
make[1]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/src'
Makefile:110: recipe for target 'sub-src-install_subtargets-ordered' failed
make: *** [sub-src-install_subtargets-ordered] Error 2
kz@klemenHp:/mnt/Extra10Gb/git/QtQuickVcp$ 

```

sirop, 2016-01-14:
```
install -m 755 -p "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/i386-linux-gnu/libmachinetalk-protobuf.so.1.0.0"
install: cannot create regular file ‘/usr/lib/i386-linux-gnu/libmachinetalk-protobuf.so.1.0.0’: Permission denied
```
then try:
```
sudo make install
```

strahlex, 2016-01-14:
It would be a lot easier if you would install the Qt SDK from qt.io instead of the one that is in the repository which probably we give you bad results anyway (https://github.com/strahlex/QtQuickVcp/issues/40). Or use the vagrant VM or the provisioning scripts: https://github.com/strahlex/machinekit-vagrant

zhivko, 2016-01-15:
OK I tried today second time in Qt Creator with the details in your wiki for project.
I got following error:
```
02:04:54: Running steps for project QtQuickVcp...
02:04:54: Configuration unchanged, skipping qmake step.
02:04:54: Starting: "/usr/bin/make" 
cd 3rdparty/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/3rdparty/3rdparty.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[1]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/3rdparty'
cd machinetalk-protobuf-qt/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/3rdparty/machinetalk-protobuf-qt/machinetalk-protobuf-qt.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/3rdparty/machinetalk-protobuf-qt'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/3rdparty/machinetalk-protobuf-qt'
make[1]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/3rdparty'
cd src/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/src.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[1]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src'
cd halremote/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/halremote/halremote.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/halremote'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/halremote'
cd controls/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/controls/controls.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/controls'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/controls'
cd halremotecontrols/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/halremotecontrols/halremotecontrols.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/halremotecontrols'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/halremotecontrols'
cd videoview/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/videoview/videoview.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/videoview'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/videoview'
cd pathview/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/pathview/pathview.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/pathview'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/pathview'
cd service/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/service/service.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/service'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/service'
cd application/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/application/application.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debug -o Makefile ) && /usr/bin/make -f Makefile 
make[2]: Entering directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/application'
make[2]: Nothing to be done for 'first'.
make[2]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src/application'
cd applicationcontrols/ && ( test -e Makefile || /home/kz/Qt/5.5/gcc/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/applicationcontrols/applicationcontrols.pro -spec linux-g++-32 CONFIG+=debug CONFIG+=declarative_debug CONFIG+=qml_debu [... cut ...]
make[1]: Leaving directory '/mnt/Extra10Gb/git/build-QtQuickVcp-Desktop_Qt_5_5_1_GCC_32bit-Debug/src'
Makefile:110: recipe for target 'sub-src-install_subtargets-ordered' failed
make[1]: *** [sub-halremote-install_subtargets-ordered] Error 2
make: *** [sub-src-install_subtargets-ordered] Error 2
02:04:56: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project QtQuickVcp (kit: Desktop Qt 5.5.1 GCC 32bit)
When executing step "Make"
02:04:56: Elapsed time: 00:02.
```

Ughhh.. don't have luck with this qt environment...

zhivko, 2016-01-15:
@sirop I tried what you suggested, but:
```
kz@klemenHp:~/git/QtQuickVcp$ sudo make install 
[sudo] password for kz: 
cd 3rdparty/ && ( test -e Makefile || /usr/lib/i386-linux-gnu/qt5/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/3rdparty/3rdparty.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory '/mnt/Extra10Gb/git/QtQuickVcp/3rdparty'
cd machinetalk-protobuf-qt/ && ( test -e Makefile || /usr/lib/i386-linux-gnu/qt5/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/3rdparty/machinetalk-protobuf-qt/machinetalk-protobuf-qt.pro -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/mnt/Extra10Gb/git/QtQuickVcp/3rdparty/machinetalk-protobuf-qt'
install -m 755 -p "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/i386-linux-gnu/libmachinetalk-protobuf.so.1.0.0"
strip --strip-unneeded "/usr/lib/i386-linux-gnu/libmachinetalk-protobuf.so.1.0.0"
ln -f -s "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/i386-linux-gnu/libmachinetalk-protobuf.so"
ln -f -s "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/i386-linux-gnu/libmachinetalk-protobuf.so.1"
ln -f -s "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/i386-linux-gnu/libmachinetalk-protobuf.so.1.0"
make[2]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/3rdparty/machinetalk-protobuf-qt'
make[1]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/3rdparty'
cd src/ && ( test -e Makefile || /usr/lib/i386-linux-gnu/qt5/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/src.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory '/mnt/Extra10Gb/git/QtQuickVcp/src'
cd halremote/ && ( test -e Makefile || /usr/lib/i386-linux-gnu/qt5/bin/qmake /mnt/Extra10Gb/git/QtQuickVcp/src/halremote/halremote.pro -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/mnt/Extra10Gb/git/QtQuickVcp/src/halremote'
/usr/lib/i386-linux-gnu/qt5/bin/qmlplugindump -nonrelocatable Machinekit.HalRemote 1.0 /mnt/Extra10Gb/git/QtQuickVcp/src/halremote/../../imports/ > /mnt/Extra10Gb/git/QtQuickVcp/src/halremote/../../imports/Machinekit/HalRemote/plugins.qmltypes
/bin/sh: 1: /usr/lib/i386-linux-gnu/qt5/bin/qmlplugindump: not found
Makefile:653: recipe for target 'install_dumppluginqmltypes' failed
make[2]: *** [install_dumppluginqmltypes] Error 127
make[2]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/src/halremote'
Makefile:73: recipe for target 'sub-halremote-install_subtargets-ordered' failed
make[1]: *** [sub-halremote-install_subtargets-ordered] Error 2
make[1]: Leaving directory '/mnt/Extra10Gb/git/QtQuickVcp/src'
Makefile:110: recipe for target 'sub-src-install_subtargets-ordered' failed
make: *** [sub-src-install_subtargets-ordered] Error 2
```
another error probably because of not correct qt environment. I will be OK if I would suceed using QT creator - it seems it has nice UI for debuging machientalk messages....
I still cannot execute MDI commands - ant want to somehow enable tracing to debug messages from "trusted" client.
If I run machinekit with -d and debug enabled in ini file, that messages gets printed to console but in some weird characters - it would probably be better to output hex of binary message (but this is maybe issue topic for machinekit project)

strahlex, 2016-01-15:
Quick guess: How do you encode your MDI commands? Machinekit only supports ascii caracters it might be that the Java environment encodes in UTF-8.

zhivko, 2016-01-15:
You have the point here! I see google protobuf has:
ByteString.copyFromUtf8(command)
I will try to use this and report back how it went. Thanks.

closed 2016-01-28 by zhivko

Willib, 2016-05-11:
Hi, I have the same question. when I build my file TTZipFileOperated.pro to generate a lib in Qt Creator, but  it output the error: " error: [../../../../debug/libTTZipFileOperated.so.1.0.0] Error 1 (ignored)". Could you help me? Thanks!

zhivko, 2016-05-11:
I moved away from QtQuickVcp, and I don't run it anymore. However I still using it as reference project to better understand, how machinetalk calls from client is supposed to work. Maybe @strahlex can help.

strahlex, 2016-05-11:
@Willib How does this question relate to QtQuickVcp?

Willib, 2016-05-12:
@strahlex oh, I am sorry, I solved my question. they really have no relations.囧...