machinekit/QtQuickVcp #47: Cetus and QtQuickVcp

hekav, 2016-02-19:
Hello,

actually I have two problems:

1. I Have a working CanOpen DS402 application running on my BeagleBone. I configured Cetus on the BeagleBone. The machinekit client works perfect except the jog stepping mode.  I runs as long as I press the plus or minus button instead of moving for a certain distance.

2. I have created a HalApplication for my Ds402 led's an it works perfectly:
[image]

strahlex, 2016-02-20:
Wow, good to see that someone is using QtQuickVcp. I think accidentally broke the jog button code in the current master some time ago (have the same problem in Machineface). I will try to fix this problem next week to get the current MachinekitClient client working correctly again.

strahlex, 2016-02-24:
The jog problem should be fixed now.

strahlex, 2016-03-21:
@hekav Can you please verify that the problem has been fixed.

hekav, 2016-03-21:
Sorry I didn't know you want a response.

Yes it works perfectly. Thanks for the Update.

closed 2016-03-21 by strahlex