machinekit/QtQuickVcp #48: Oops I pressed the Submit Button too early

hekav, 2016-02-19:
After that was working I integrated my Panel in Cetus an deployed the new Version to my BeagleBone. This works also: 
[image]

I'm not able to bring the signals from my hal component (controlDs402 from above)  into my panel in cetus. 
Is this possible and how can it be done ?
   
Regards
Helmut

dschbrs, 2016-02-19:
Maybe you need add in Cetus.qml:

                YouNewTabName {
                     active: true
         }

hekav, 2016-02-19:
That's what I did not in Cetus.qml but in DisplayPanel.qml

TabView {
    width: 100
    height: 62

    /*PrinterTab {}
    ErrorTab {}*/
    Tab {
        title: qsTr("Configuration")
        active: true

        Item {
            ConfigurationPanel {
                anchors.fill: parent
            }                 
        }
     }
    Tab {
        title: qsTr("DS402")
        active: true

        Item {

            DS402Panel {
                anchors.fill: parent
            }
        }
     }
}

dschbrs, 2016-02-19:
This worked for me, after trying days :

In Cetus.qml

    Service {
        id: halrcompService
        type: "halrcomp"
    }

    Service {
        id: halrcmdService
        type: "halrcmd"
    }

    HalRemoteComponent {
        id: halRemoteComponent
        halrcmdUri: halrcmdService.uri
        halrcompUri: halrcompService.uri
        ready: (halrcmdService.ready && halrcompService.ready) || connected
        name: "yourcompnamel"
        containerItem: container
        create: false
        onErrorStringChanged: console.log(errorString)
    }

....

TabView {

...

              MiscTab {
                     active: true
         }
 }

In the MiscTab.qml

import Machinekit.HalRemote 1.0
import Machinekit.HalRemote.Controls 1.0
import Machinekit.Service 1.0

a led example:

            HalPin {
                id: led_resetservo
                name: "resetservo_led"
                type: HalPin.Bit
                direction: HalPin.In
            }

hekav, 2016-02-19:
This did the trick !

Thanks a lot

strahlex, 2016-02-20:
Yes, the HalApplicationWindow does create the HalRemoteComponent for you. If you use it inside another UI you have to add it manually.

closed 2016-02-20 by strahlex