machinekit/QtQuickVcp #51: Qt 5.6.0 - error during "sudo make install"

koppi, 2016-04-19:
With Qt 5.6.0 x86 from qt.io on Debian/Jessie during ```sudo make install``` I'm getting this error:
```bash
koppi@x200:/data/QtQuickVcp$ sudo make install
cd 3rdparty/ && ( test -e Makefile || /usr/bin/qmake /data/QtQuickVcp/3rdparty/3rdparty.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory '/data/QtQuickVcp/3rdparty'
cd machinetalk-protobuf-qt/ && ( test -e Makefile || /usr/bin/qmake /data/QtQuickVcp/3rdparty/machinetalk-protobuf-qt/machinetalk-protobuf-qt.pro
-o Makefile ) && make -f Makefile install
make[2]: Entering directory '/data/QtQuickVcp/3rdparty/machinetalk-protobuf-qt'
install -m 755 -p "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/x86_64-linux-gnu/libmachinetalk-protobuf.so.1.0.0"
strip --strip-unneeded "/usr/lib/x86_64-linux-gnu/libmachinetalk-protobuf.so.1.0.0"                                                               |
ln -f -s "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/x86_64-linux-gnu/libmachinetalk-protobuf.so"                                                |
ln -f -s "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/x86_64-linux-gnu/libmachinetalk-protobuf.so.1"                                              |
ln -f -s "libmachinetalk-protobuf.so.1.0.0" "/usr/lib/x86_64-linux-gnu/libmachinetalk-protobuf.so.1.0"                                            |
make[2]: Leaving directory '/data/QtQuickVcp/3rdparty/machinetalk-protobuf-qt'                                                                    |
make[1]: Leaving directory '/data/QtQuickVcp/3rdparty'                                                                                            |
cd src/ && ( test -e Makefile || /usr/bin/qmake /data/QtQuickVcp/src/src.pro -o Makefile ) && make -f Makefile install                            |
make[1]: Entering directory '/data/QtQuickVcp/src'                                                                                                |
cd halremote/ && ( test -e Makefile || /usr/bin/qmake /data/QtQuickVcp/src/halremote/halremote.pro -o Makefile ) && make -f Makefile install      |
make[2]: Entering directory '/data/QtQuickVcp/src/halremote'                                                                                      |
/usr/bin/qmlplugindump -nonrelocatable Machinekit.HalRemote 1.0 /data/QtQuickVcp/src/halremote/../../imports/ > /data/QtQuickVcp/src/halremote/../|
../imports/Machinekit/HalRemote/plugins.qmltypes                                                                                                  |
QDeclarativeComponent: Component is not ready                                                                                                     |
"file:///data/QtQuickVcp/imports/typelist.qml:2:1: a component declaration requires 3 arguments, but 2 were provided"                             |
Makefile:633: recipe for target 'install_dumppluginqmltypes' failed                                                                               |
make[2]: *** [install_dumppluginqmltypes] Error 3                                                                                                 |
make[2]: Leaving directory '/data/QtQuickVcp/src/halremote'                                                                                       |
Makefile:69: recipe for target 'sub-halremote-install_subtargets-ordered' failed                                                                  |
make[1]: *** [sub-halremote-install_subtargets-ordered] Error 2                                                                                   |
make[1]: Leaving directory '/data/QtQuickVcp/src'                                                                                                 |
Makefile:106: recipe for target 'sub-src-install_subtargets-ordered' failed                                                                       |
make: *** [sub-src-install_subtargets-ordered] Error 2                                                                                            |
koppi@x200:/data/QtQuickVcp$                                                                                                                      |
--------------------------------------------------------------------------------------------------------------------------------------------------

```
Any ideas?

strahlex, 2016-04-19:
I have never tried make install with sudo. It looks like the qmlplugindumps application causes problems in this scenario. Solution: Comment the qmlplugindump related lines in the `deployment.pri` https://github.com/strahlex/QtQuickVcp/blob/master/src/deployment.pri#L58

I have to reevaluate if this "hack" is still necessary to get the QML Designer preview.

yishinli, 2016-05-29:
I had the same problem while building QtQuickVcp with Qt5.6. Here's my workaround:
`sudo ln -sf /opt/Qt5.6.0/Tools/QtCreator/share/qtcreator/qmldesigner/propertyEditorQmlSources/HelperWidgets/ /opt/Qt5.6.0/5.6/gcc_64/qml`

Verify if Machinekit.HalRemote is ready:
`qmlplugindump -v Machinekit.HalRemote 1.0 imports/`

koppi, 2016-05-29:
@yishinli Thank's for this info. I just tried this, and it did the trick.

closed 2016-05-29 by koppi