machinekit/QtQuickVcp #53: Unit testing infrastructure for QtQuickVcp

yishinli, 2016-05-31:
There is a tests/SpeedTest/ directory in QtQuickVcp. However, after compiling it, we could not figure out how to run it.

I'm thinking of creating a unit testing infrastructure for QtQuickVcp. Please advice the possible approaches.

strahlex, 2016-05-31:
The SpeedTest is just an application to test the performance of QtQuickVcp under different network conditions. There are currently no unit tests in QtQuickVcp. I recommend testing of UI components using the QtQuick Test suite: http://doc.qt.io/qt-5/qttest-qmlmodule.html
QTest can be applied to test the C++ modules: http://doc.qt.io/qt-5/qtest-overview.html
Additionally integration testing of the network modules would be suitable. I am also working on formally verifying the Machinetalk protocols and code generation for the state machine code.

strahlex, 2016-07-20:
I have added a basic testing infrastructure for QML https://github.com/strahlex/QtQuickVcp/tree/master/tests. C++ tests can be added in Qt Create by adding a new Autotest subproject.

machinekoder, 2017-06-23:
@yishinli Any more questions?

yishinli, 2017-06-24:
Please advise how to run and inspect the results for halremotetests/ and qmltests/.

machinekoder, 2017-10-02:
@yishinli Just compile and run the halremotetests and qmltests executables.

machinekoder, 2017-10-02:
In the meantime I have moved to Catch as unit testing framework: https://github.com/philsquared/Catch

closed 2017-11-01 by machinekoder