machinekit/QtQuickVcp #54: Qt5.5.1 on Raspberry Pi2: failed to sudo make install

ddtrng, 2016-06-06:
Hello, 
With Raspbian Qt5.5.1: Linux pi4 4.4.8-v7+ #880 SMP Fri Apr 22 21:55:04 BST 2016 armv7l GNU/Linux

pi@pi4:~/proj/remote-ui/QtQuickVcp.master(master)$ qtchooser -print-env
```
QT_SELECT="default"
QTTOOLDIR="/usr/lib/arm-linux-gnueabihf/qt5/bin"
QTLIBDIR="/usr/lib/arm-linux-gnueabihf"
```
make runs whi no errors.
durring sudo make install, getting this error

pi@pi4:~/proj/remote-ui/QtQuickVcp.master(master)$ sudo make install
```
cd 3rdparty/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty/3rdparty.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty'
cd machinetalk-protobuf-qt/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty/machinetalk-protobuf-qt/machinetalk-protobuf-qt.pro -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty/machinetalk-protobuf-qt'
install -m 755 -p libmachinetalk-protobuf.so.1.0.0 /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so.1.0.0
strip --strip-unneeded /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so.1.0.0
ln -f -s libmachinetalk-protobuf.so.1.0.0 /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so
ln -f -s libmachinetalk-protobuf.so.1.0.0 /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so.1
ln -f -s libmachinetalk-protobuf.so.1.0.0 /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so.1.0
make[2]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty/machinetalk-protobuf-qt'
make[1]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty'
cd src/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/src/src.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src'
cd halremote/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote/halremote.pro -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote'
/usr/lib/arm-linux-gnueabihf/qt5/bin/qmlplugindump -nonrelocatable Machinekit.HalRemote 1.0 /home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote/../../imports/ > /home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote/../../imports/Machinekit/HalRemote/plugins.qmltypes
QQmlComponent: Component is not ready
Makefile:518: recipe for target 'install_dumppluginqmltypes' failed
make[2]: *** [install_dumppluginqmltypes] Error 3
make[2]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote'
Makefile:73: recipe for target 'sub-halremote-install_subtargets-ordered' failed
make[1]: *** [sub-halremote-install_subtargets-ordered] Error 2
make[1]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src'
Makefile:110: recipe for target 'sub-src-install_subtargets-ordered' failed
make: *** [sub-src-install_subtargets-ordered] Error 2
```
after that, I try to  comment the qmlplugindump related lines in the deployment.pri [https://github.com/strahlex/QtQuickVcp/blob/master/src/deployment.pri#L58](https://github.com/strahlex/QtQuickVcp/blob/master/src/deployment.pri)

pi@pi4:~/proj/remote-ui/QtQuickVcp.master(master)$ sudo make install
```
cd 3rdparty/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty/3rdparty.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty'
cd machinetalk-protobuf-qt/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty/machinetalk-protobuf-qt/machinetalk-protobuf-qt.pro -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty/machinetalk-protobuf-qt'
install -m 755 -p libmachinetalk-protobuf.so.1.0.0 /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so.1.0.0
strip --strip-unneeded /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so.1.0.0
ln -f -s libmachinetalk-protobuf.so.1.0.0 /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so
ln -f -s libmachinetalk-protobuf.so.1.0.0 /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so.1
ln -f -s libmachinetalk-protobuf.so.1.0.0 /usr/lib/arm-linux-gnueabihf/libmachinetalk-protobuf.so.1.0
make[2]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty/machinetalk-protobuf-qt'
make[1]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/3rdparty'
cd src/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/src/src.pro -o Makefile ) && make -f Makefile install
make[1]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src'
cd halremote/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote/halremote.pro -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote'
/usr/lib/arm-linux-gnueabihf/qt5/bin/qmake -o Makefile halremote.pro
mkdir -p /usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/HalRemote/ 
cp -f /home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote/../../imports/Machinekit/HalRemote/plugins.qmltypes /usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/HalRemote/plugins.qmltypes
install -m 644 -p /home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote/qmldir /usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/HalRemote/
install -m 755 -p ../../imports/Machinekit/HalRemote/libmachinekithalremoteplugin.so /usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/HalRemote/libmachinekithalremoteplugin.so
strip --strip-unneeded /usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/HalRemote/libmachinekithalremoteplugin.so
make[2]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src/halremote'
cd controls/ && ( test -e Makefile || /usr/lib/arm-linux-gnueabihf/qt5/bin/qmake /home/pi/proj/remote-ui/QtQuickVcp.master/src/controls/controls.pro -o Makefile ) && make -f Makefile install
make[2]: Entering directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src/controls'
/usr/lib/arm-linux-gnueabihf/qt5/bin/qmake -o Makefile controls.pro
mkdir -p /usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/Controls/ 
cp -f /home/pi/proj/remote-ui/QtQuickVcp.master/src/controls/../../imports/Machinekit/Controls/plugins.qmltypes /usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/Controls/plugins.qmltypes
cp: cannot stat ‘/home/pi/proj/remote-ui/QtQuickVcp.master/src/controls/../../imports/Machinekit/Controls/plugins.qmltypes’: No such file or directory
Makefile:578: recipe for target 'install_copypluginqmltypes' failed
make[2]: *** [install_copypluginqmltypes] Error 1
make[2]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src/controls'
Makefile:116: recipe for target 'sub-controls-install_subtargets-ordered' failed
make[1]: *** [sub-controls-install_subtargets-ordered] Error 2
make[1]: Leaving directory '/home/pi/proj/remote-ui/QtQuickVcp.master/src'
Makefile:110: recipe for target 'sub-src-install_subtargets-ordered' failed
make: *** [sub-src-install_subtargets-ordered] Error 2
```
Did I miss something package?

ddtrng, 2016-06-06:
after that, I comment the qmlplugindump related lines in the deployment.pri [https://github.com/strahlex/QtQuickVcp/blob/master/src/deployment.pri#L58 and #L67](https://github.com/strahlex/QtQuickVcp/blob/master/src/deployment.pri)
make -j2 && sudo make install && make docs && sudo make install_docs
no errors in this time.
pi@pi4:~/proj/remote-ui/MachinekitClient(master)$ qmake -r
pi@pi4:~/proj/remote-ui/MachinekitClient(master)$ make
```
g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIE -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_QML_LIB -DQT_WIDGETS_LIB -DQT_NETWORK_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtQuick -isystem /usr/include/arm-linux-gnueabihf/qt5/QtQml -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -o main.o main.cpp
/usr/lib/arm-linux-gnueabihf/qt5/bin/rcc -name qml qml.qrc -o qrc_qml.cpp
g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIE -DQT_NO_DEBUG -DQT_QUICK_LIB -DQT_QML_LIB -DQT_WIDGETS_LIB -DQT_NETWORK_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -I. -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtQuick -isystem /usr/include/arm-linux-gnueabihf/qt5/QtQml -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -I. -o qrc_qml.o qrc_qml.cpp
g++ -Wl,-O1 -o machinekit-client main.o qrc_qml.o   -lQt5Quick -lQt5Qml -lQt5Widgets -lQt5Network -lQt5Gui -lQt5Core -lGLESv2 -lpthread 
```
pi@pi4:~/proj/remote-ui/MachinekitClient(master)$ ./machinekit-client 
```
libEGL warning: DRI2: failed to authenticate
qml: Starting to load the main application.
qrc:///main.qml:32:1: a component declaration requires two or three arguments, but 1 were provided
file:///usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/Controls/qmldir:8: unexpected '.'
file:///usr/lib/arm-linux-gnueabihf/qt5/qml/Machinekit/Controls/qmldir:11: unexpected '.'
qml: Finishing the splash screen progress bar.
qml: Hiding the splash screen.
qml: Deleting the splash screen.
```
Did I miss something package or can not comment  two lines(deployment.pri#L58 and #L67)

koppi, 2016-06-13:
See https://github.com/strahlex/QtQuickVcp/issues/51

ddtrng, 2016-06-14:
Use [Native Build of Qt 5.4.1 on a Raspberry Pi](https://wiki.qt.io/Native_Build_of_Qt_5.4.1_on_a_Raspberry_Pi), Problem is solved

closed 2016-06-14 by ddtrng