machinekit/QtQuickVcp #55: ApplicationCore.qml:149: Binding loop detected for property "ready"

yishinli, 2016-06-21:
After doing "HomeAll", the MachinekitClient became sluggish, which took 105% of CPU loading, and dump the following warning message:
`file:///usr/local/qt5.6.2/qml/Machinekit/Application/ApplicationCore.qml:149:5: QML ApplicationError: Binding loop detected for property "ready"`

Platform: Raspberry Pi 3, Raspbian Jessie, Native built Qt 5.6 (git version)

Please advise what would be the potential causes for "Binding loop" of "ready" on [ApplicationCore.qml:149](https://github.com/strahlex/QtQuickVcp/blob/master/src/application/ApplicationCore.qml#L149).

strahlex, 2016-06-26:
The Binding loops are usually caused when the client state changes too "quickly" depending on the ready signal. This happens because the .connected property is also back-fed to the ready signal. Usually this warning message can be safely ignored.

The performance problems can be best backtraced using the QML profiler which is part of Qt Creator. Here you can find more information: http://doc.qt.io/qtcreator/creator-qml-performance-monitor.html
Not sure if the QML profiler wroks with remote processes. However, you can try to reproduce the problem on a platform where Qt Creator is available.

yishinli, 2016-06-26:
@strahlex , you are correct; that binding loop message can be ignored. The real problem is "the GUI freezing", which is caused by QApplicationStatus::statusMessageReceived().

My current approach is moving statusMessageReceived() to another thread whit QtConcurrent::run(), and executing QEventLoop while processing the received message. The GUI is no longer freezing. The revised code is [here](https://github.com/araisrobo/QtQuickVcp/commits/concurrent).

However, this approach might not be efficient as the thread is blocked while processing statusMessageReceived(). I've observed burst of statusMessageReceived. As there are 4 cpu cores on Raspberry Pi 3, please advise how could we take benefit of those cores by parallel processing the received messages?

strahlex, 2016-06-26:
This is interesting. Can you try to find out what part of handling the message causes the GUI freeze? Does this happen when a new client connects or always?

My guess is that the inefficient code is either handling the status message and putting it into the JSON structure or passing it to QML. https://github.com/strahlex/QtQuickVcp/blob/master/src/common/service.cpp#L108

yishinli, 2017-01-30:
The performance issue is addressed in [PR67](https://github.com/machinekoder/QtQuickVcp/pull/67).

closed 2017-01-30 by yishinli