machinekit/QtQuickVcp #57: Feature Request -- GCode Editing

yishinli, 2016-07-13:
I'm thinking of implementing GCode Editing function for QtQuickVcp. So that we may edit current opened NGC file.

Would it be reasonable to revise [QGCodeProgramModel](https://github.com/strahlex/QtQuickVcp/blob/master/src/pathview/qgcodeprogrammodel.cpp), and implement gcode-editing function in it?

I'm not familiar with the QtQuickVcp architecture yet, and am tracing the code. Any help are appreciated.

-Yishin

yishinli, 2016-07-13:
Is [Code Editor Example](http://doc.qt.io/qt-5/qtwidgets-widgets-codeeditor-example.html) a suitable template for NGC editor?

strahlex, 2016-07-13:
This example helps only partially since it is for QWidgets and not QtQuick. I have found this here https://forum.qt.io/topic/36130/example-of-textedit-using-a-qsyntaxhighlighter including a link to a QML TextEdit implementation for Python Syntax highlighting. I think this should be a good starting point to get GCode highlighting. I recommend starting by getting this up and running I can help you with the GCodeModel.

QtQuickVcp uses a model-view approach for the GCode, meaning loading and previewing both as text and in the 3D preview. Saving the GCode from the model should be trivial, similar as GCodeLoader does the loading. Editing requires modification of the model. The SourceView component currently only displays GCode. It has to be extended with editing capabilities, the GCodeModel itself should already be capable of this.

yishinli, 2016-07-14:
I'm trying to understand your suggestion correctly:
1. SourceView.qml: modify ListView to support GCode Editing.
2. GCodeModel: to implement "insert line", "delete line", "undo", "redo", "save file"
3. With this approach, we may get 3D preview for each newly updated lines.

The other approach:
1. SourceView.qml: switch from ListView to multiple lines TextEdit to edit GCode.
2. SourceView.qml: switch back to ListView after saving GCode.
3. This approach would be easier; however, it can not get 3D preview while editing GCode lines.

Is my understand correct?
And, which approach would be better for GCode Editing?

strahlex, 2016-07-14:
No you would not get instant preview in any case since this requires the preview interpreter to be executed. The seconds approach seems more feasible since you only need to edit the GCode and upload it again after editing.

yishinli, 2016-07-16:
In SourceView.qml, there is a PathViewObject, which deals with gcodeProgramLoader and gcodeProgramModel.
Question:
How does gcodeProgramLoader.load() get called after opening a file with OpenAction.qml?

yishinli, 2016-07-16:
1. add an EditProgramAction to toggle Edit-mode signal
2. In SourceVeiw, switch between ScrollView and TextEdit while Edit-mode is toggled.
However, I have problem creating the Edit-mode signal between EditProgramAction and SourceView.

Please advice if this approach is suitable, and how to create the Edit-mode signal?

yishinli, 2016-07-17:
[Best way to have qml function and c++ slot and vice versa for the same item](http://stackoverflow.com/questions/20988224/best-way-to-have-qml-function-and-c-slot-and-vice-versa-for-the-same-item)
This link seems to be the answer for connecting Edit-mode signal between QML and C++ model.

yishinli, 2016-07-17:
WIP: gcodeEdieMode property was implemented; 
git repositories: [QtQuickVcp](https://github.com/araisrobo/QtQuickVcp/tree/gcodeEdit), [Machineface](https://github.com/araisrobo/Machineface)
[image]
TODO:
1. Update the content of gcodeEdit with gcodeProgramModel
2. Implement required actions(copy, past, save, etc.) for gcodeEdit
3. Syntax highlight for gcodeEdit

yishinli, 2016-07-18:
WIP: Save file after toggling gcodeEditMode. 
git repositories: [QtQuickVcp](https://github.com/araisrobo/QtQuickVcp/tree/gcodeEdit), [Machineface](https://github.com/araisrobo/Machineface)
TODO:
1. Implement actions(select, copy, past, ... etc.) for gcodeEdit
2. Syntax highlight for gcodeEdit

strahlex, 2016-07-20:
I think copy and paste are provided by the TextEdit itself, you just need to add some buttons/context menu. However, GCodeProgramLoader needs a save function to dump the contents of the GCodeModel into a file.

`gcodeProgramLoader.load` is triggered in the [PathViewCore](https://github.com/strahlex/QtQuickVcp/blob/master/src/pathview/PathViewCore.qml#L67)

yishinli, 2016-07-21:
> GCodeProgramLoader needs a save function to dump the contents of the GCodeModel into a file
In GCodeProgramLoader, there is a[ text property which was set](https://github.com/araisrobo/QtQuickVcp/blob/gcodeEdit/src/pathview/qgcodeprogramloader.cpp#L82) before constructing the GCodeModel.

strahlex, 2016-07-29:
This here popped up on my radar: https://github.com/ArcEye/QGCodePreview

yishinli, 2016-07-30:
Also, please check GCoder - [https://youtu.be/9D3hMXP5-QM](https://youtu.be/9D3hMXP5-QM)

machinekoder, 2017-10-01:
@yishinli Does the desktop editing feature work fo you?

machinekoder, 2017-11-24:
I close this issue for now since the feature is provided by desktop editing.

closed 2017-11-24 by machinekoder