machinekit/QtQuickVcp #59: Service discovery Android 5.1

sirop, 2016-07-14:
_Do not know if the issue is more about pymachinetalk or QtQuickVcp,
but as it can overlap with https://github.com/strahlex/QtQuickVcp/issues/34 , I'll put it here._

How to reproduce:
```
git clone https://github.com/machinekit/pymachinetalk
cd pymachinetalk
wget https://bootstrap.pypa.io/ez_setup.py -O - | sudo  python
python setup.py build_py
```

```
mklauncher -d examples/basic/anddemo.py
announcing mklauncher

service: dsname = tcp://%(fqdn)s:64877 port = 64877 txtrec = ['dsn=tcp://%(fqdn)s:64877', 'uuid=a42c8c6b-4025-4f83-ba28-dad21114744a', 'instance=5fe63546-4999-11e6-8166-30b5c21f5050', 'service=launcher'] name = Machinekit Launcher on %(fqdn)s
service: dsname = tcp://%(fqdn)s:54042 port = 54042 txtrec = ['dsn=tcp://%(fqdn)s:54042', 'uuid=a42c8c6b-4025-4f83-ba28-dad21114744a', 'instance=5fe63547-4999-11e6-8166-30b5c21f5050', 'service=launchercmd'] name = Launchercmd service on %(fqdn)s pid 6322
```
This gives this picture:
[image]

Clicking on the instance yileds:
```
sending launcher message
process command called, id: ['launcher-3321']
sending launcher message
process command called, id: ['launcher-3321']
sending launcher message
process command called, id: ['launcher-3321']
sending launcher message
process command called, id: ['launcher-3321']
sending launcher message
process command called, id: ['launcher-3321']
sending launcher message
sending launcher message
process command called, id: ['launcher-3321']
sending launcher message
process command called, id: ['launcher-3321']
sending launcher message
process command called, id: ['launcher-3321']
sending launcher message
process command called, id: ['launcher-3321']
```
with this pic
[image]

What do I miss?

Machinekit runs on:
```
$ uname -a
Linux debian-master 3.8-1-xenomai.x86-amd64 #1 SMP Debian 3.8.13-12~1jessie~1da x86_64 GNU/Linux
~/machinekit$ git log | head -10
commit 0766e79cd9597c7786ea415cfcfd2862ca3e68f5
Merge: 722247c 99b4b36
Author: Michael Haberler <mail17@mah.priv.at>
Date:   Wed Jul 13 19:56:59 2016 +0200

    Merge pull request #976 from strahlex/hal-graph

    DO NOT MERGE: scripts: add hal-graph tool

commit 722247ca0c0077c790bdc3ac9eba16f5bd56dfa3
```

Android runs on:
[image]

strahlex, 2016-07-19:
mklauncher needs to be started with a folder containing a `launcher.ini` file as argument, e.g.:
`mklauncher -d examples/basic/` in your case simply remove the `anddemo.py`

sirop, 2016-07-19:
Indeed. If I'd have used 
```
git clone https://github.com/strahlex/anddemo
cd anddemo
mklauncher -d .
```
then it works.

Tested on Android 5.1.1 and Android 4.4.2

closed 2016-07-19 by sirop