machinekit/QtQuickVcp #127: Problem on Windows 10

matatasb, 2017-01-13:
Hello,
First of all, I want to thank you all of you for your great job on MK and QtQuickVcp.
I´ve installed via Vagrant the MachineKit SDK that Alexander made on two differents PCs. In one I could connect the Client with the server running on the virtal machine on Windows. Everything worked OK. 
In another installation I tried to do the same thing and I couldn´t connect. Later I tried to connect with the annddemo on BB but after I run the .py the client on Windows closed.

Is there any log file or something I could check or a way I could debug myself?

If I must, I could install Qt and try to debug the error. I don't have much experience, It will take more time, but I could try if there isn't other way.

Thank you all, Matias

machinekoder, 2017-01-17:
You can start the client from the command line to see the debug output and potential error messages.

On Windows, there is the additional issue of choosing the graphic rendering engine. Can you please to set the `QT_ANGLE_PLATFORM` variable manually as mentioned here: http://stackoverflow.com/questions/27704529/deploying-qt5-on-windows-without-hardware-acceleration

For this purpose, you need to create a Batch script (.bat file). You can add pause at the end of the file to keep the window open or run it directly from the command line (Shift-Right-Click > Open command shell here in Windows Explorer.

Debugging on Windows is a little bit more complicated. You need to install the Windows SDK (VS Studio Express/Community) and a few more things. If you want to try it please take a look the README file.

matatasb, 2017-01-20:
Hello Alex,
Thank you for your response. I tried what you told me but I the console didn't log anything. I checked the firewall, uninstalled the anti-virus. Is there any command I could add like -d to to debug? Just after I start the server on the BB the program crush.
I tried what you said about QT_ANGLE_PLATFORM but I don't feel confident with that... 
I followed the README instruction to compile and run QtQuickVCP on Qt. I'm still finding some troubles on that, but I want to try a few more things before I bother someone or you with that issue. 
If you have any other idea, I will give it a try.
Thank you, Matias

machinekoder, 2017-01-22:
Have you tried creating a batch script setting the `QT_ANGLE_PLATFORM` environment variable?
Suitable values are:
* d3d11: Use Direct3D 11
* d3d9: Use Direct3D 9
* warp: Use the Direct3D 11 software rasterizer
See also:
http://ss64.com/nt/set.html

```
SET QT_ANGLE_PLATFORM=d3d11
machinekit-client.exe
```

matatasb, 2017-01-22:
Hello, thank you again for your support. I tried what you said, but there wasn't any change. I'm trying with the simplest example (anddenmo).
I ran the command eventvwr and give the next message in the log:
****
Faulting application name: machinekit-client.exe, version: 0.0.0.0, time stamp: 0x587ca0dc
Faulting module name: machinekitserviceplugin.dll, version: 0.0.0.0, time stamp: 0x587ca0aa
Exception code: 0xc0000005
Fault offset: 0x00011cc4
Faulting process ID: 0x1b44
Faulting application start time: 0x01d27502baefbcde
Faulting application path: C:\Users\matias\Desktop\MachinekitClient_Development-201701161026-master-7b1d557-x86\MachinekitClient\machinekit-client.exe
Faulting module path: C:\Users\matias\Desktop\MachinekitClient_Development-201701161026-master-7b1d557-x86\MachinekitClient\Machinekit\Service\machinekitserviceplugin.dll
Report ID: be2c345b-c2e7-4474-b8c1-cf63009f1bb1
Faulting package full name: 
Faulting package-relative application ID: 
***
Fault bucket 108700548404, type 1
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: machinekit-client.exe
P2: 0.0.0.0
P3: 587ca0dc
P4: machinekitserviceplugin.dll
P5: 0.0.0.0
P6: 587ca0aa
P7: c0000005
P8: 00011cc4
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER32CE.tmp.WERInternalMetadata.xml
\\?\C:\Users\matias\AppData\Local\Temp\WER3EB6.tmp.appcompat.txt
triagedump.dmp
\\?\C:\Users\matias\AppData\Local\Temp\WER407D.tmp.WERDataCollectionFailure.txt

These files may be available here:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_machinekit-clien_15518884f3b920a3991d29223980f138a17756_b18c1463_03934694
Analysis symbol: 
Rechecking for solution: 0
Report ID: be2c345b-c2e7-4474-b8c1-cf63009f1bb1
Report Status: 4104
Hashed bucket: 4770394ed093ad9baa811581b5ffa2e3

Thanks

matatasb, 2017-02-02:
Hello,
I tried to ping the Virtual Machine but I didn't have a response from it. I uninstalled 2 programs related to Ethernet conection, the only one I remember was Nmap (maybe the other was a driver or something like that of that program). 
After that everything works perfect.

Thank you for your support and the magnifiecient work that you are doing. Now I'm working on a interface based on gmocappy's plasma appereance to be used in QtQuickVCP. For me it's a long learning curve but very funny. 
If someday I feel confident enough I would like to make a mini-tutorial for QtQuickVCP and the design of GUIs.

Thank you again! You can close the issue.

machinekoder, 2017-02-02:
Nice to hear that it works now. 

I really would appreciate to see a mini-tutorial for QtQuickVcp.

closed 2017-02-02 by machinekoder