machinekit/QtQuickVcp #162: M00/M01 Resume Behavior

dkhughes, 2017-04-24:
When a program encounters an `M00` or enabled `M01`, the GUI controls do not properly reflect the state of the control to allow the operator to continue a program. The Pause/Resume action is enabled, but the user must first `Pause` then `Resume` even though machinekit only requires the resume command to be sent to continue running the program.

Here is a sample program to duplicate the behavior:
```
G17 G20 G90 G40 G54

#<count>=0.0
o100 do
(Simulate a M00 condition)
(like a manual tool change op.)
G01 X5.0 Y5.0 F150.
M00 (Wait on operator)
G01 X0. Y0.0
#<count> = [#<count> + 1.];
o100 while[#<count> LT 10.]

G40 G1 X0.0 F100.

M02
```

**What's expected**:
The `M00` should enable either the resume button or enable the RunProgramAction again.

**What happens with the RunProgramAction?**:
The running flag is still active since the program is technically running. This check [https://github.com/qtquickvcp/QtQuickVcp/blob/master/src/applicationcontrols/RunProgramAction.qml#L51](https://github.com/qtquickvcp/QtQuickVcp/blob/master/src/applicationcontrols/RunProgramAction.qml#L51) keeps the run button disabled.

**What happens with the PauseResumeAction?**:
The action continues to offer `Pause` as the only option since the GUI is unaware that the `M00` caused the control to pause already. This check [https://github.com/qtquickvcp/QtQuickVcp/blob/master/src/applicationcontrols/PauseResumeProgramAction.qml#L40](https://github.com/qtquickvcp/QtQuickVcp/blob/master/src/applicationcontrols/PauseResumeProgramAction.qml#L40] makes activating the PauseResumeAction twice required before the control continues execution.

**The proposed fix**:
I made the PauseResumeAction also dependent on the `status.task.taskPaused` field. This allowed for correctly representing the pause state whether the user initiated it or the ngc program did.

If this sounds acceptable I can submit a PR.

machinekoder, 2017-04-25:
Yes, the fix sounds good. Please go ahead and create a PR.

machinekoder, 2017-05-08:
@dkhughes Any progress on this?

machinekoder, 2017-05-09:
Good point. We may also bind the checked state to `_paused`:
Either via:
```
checked: _paused
or 
Binding { target: root; property: "checked"; value: root._paused }
```

machinekoder, 2017-05-09:
Feel free to create a PR. Makes things easier to discuss.

closed 2017-05-10 by machinekoder