machinekit/QtQuickVcp #167: Disable "debug" output remote UI

luminize, 2017-05-17:
I've made a UI and compiled it, then start it from a python program with `launcher.start_process('bash ~/path/to/application -W')

All output is shown in the terminal. (any terminal for that matter) Can I disable that somehow?

machinekoder, 2017-05-17:
Yes, compile it in release mode (QtQuickVcp + the UI).

machinekoder, 2017-05-17:
If there is still some debug output left you can pipe it somewhere else: `foo > bar.log`

luminize, 2017-05-17:
That doesn't seem to solve it. Still as much output as before.
Maybe I erred and it's not debug messages. Anyway these output (and much more) I see:
```
[ SD ] trying to open network session
[ SD ] number of configs:  3
[ SD ] network config:  "Ethernet" 1 "Wired connection 1"
[ SD ] "Initializing JDNS multicast"
[ SD ] Started query 1 "_config._sub._machinekit._tcp.local" 12
[ SD ] Started query 2 "_launcher._sub._machinekit._tcp.local" 12
...
...
[ "Remote Component Base" ] Event CONNECT
[ "Remote Component Base" ] State TRYING
[ "tubeturner" ] pin added:  "en_recipy"
[ "tubeturner" ] pin added:  "cbo_recipy"
[ "tubeturner" ] pin added:  "en_enter"
...
...
[ "Remote Component Base - halrcmd" ] Event HEARTBEAT TICK
[ "Remote Component Base - halrcmd" ] sent message "type: MT_PING\n"
[ "Remote Component Base - halrcmd" ] Event ANY MSG SENT
[ "Remote Component Base - halrcmd" ] server message "type: MT_PING_ACKNOWLEDGE\nuuid: \"(RG\\250;\\034\\021\\347\\253\\231\\000\\037)\\000\\354o\"\n"
[ "Remote Component Base - halrcmd" ] Event ANY MSG RECEIVED
...
...
```

machinekoder, 2017-05-18:
Make sure you **Rebuild** the QtQuickVcp project.

luminize, 2017-05-18:
yes, I put in release mode, then rebuild, (CTRL+B)
after that run (CTRL+R) the app and you'll see also output in the application output window (3) at the bottom in the Qt environment.

machinekoder, 2017-05-18:
`make install` was run? Because the application uses QtQuickVcp from the install path.
[image]

luminize, 2017-05-18:
apologies, I compiled only the UI in release mode. Compiling QtQuickVCP in release mode solved the problem.
Thanks @machinekoder

closed 2017-05-18 by luminize