machinekit/QtQuickVcp #168: Question about 32 bit pc's and QtQuickVCP minmal version

luminize, 2017-05-23:
I have a PC with a 32 bit installation. I noticed that the latest 32 bit Qt only goes as far as Qt 5.5 which has QtQuick 1.
Is this usable for QtQuickVCP?
Is there a minimal version needed?

machinekoder, 2017-05-23:
The Qt SDK has been dropped for 32bit Linux due to the small number of users. However, you can compile Qt yourself or use the packages from your Linux distribution.

If you have a working recipe to build Qt for 32 bit Linux then I can also integrate it into the build system. The dependencies are usually minimal: http://doc.qt.io/qt-5/linux-building.html

luminize, 2017-05-23:
ok, I guess it will be in the end easier if I install the 64 bit OS. Than I can use the binary build of my application.

Thanks

closed 2017-05-23 by luminize