machinekit/QtQuickVcp #169: MachinekitClient fails to start

bschousek, 2017-06-15:
Trying to run the Machinekit Client from an Arch Linux system yields errors:
```
~/Downloads/MachinekitClient_Development-201705181342-master-fcc665f-x64.AppImage -d

libGL error: unable to load driver: nouveau_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: nouveau
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
Unrecognized OpenGL version
Unrecognized OpenGL version

```

Googling those errors led to [https://askubuntu.com/questions/635851/error-in-installing-steam-on-ubuntu-15-04](https://askubuntu.com/questions/635851/error-in-installing-steam-on-ubuntu-15-04) which led me to  trying to force a particular version of libstdc++ as follows:

`LD_PRELOAD='/usr/$LIB/libstdc++.so.6' ~/Downloads/MachinekitClient_Development-201705181342-master-fcc665f-x64.AppImage -d `

Forcing the c++ library version worked in this instance.

For reference:
```
uname -a                                                                                                                                             
Linux frozen 4.11.3-1-ARCH #1 SMP PREEMPT Sun May 28 10:40:17 CEST 2017 x86_64 GNU/Linux

```

machinekoder, 2017-06-22:
Thanks for reporting the issue. This PR might fix the problem: https://github.com/qtquickvcp/QtQuickVcp/pull/172

propcoder, 2017-07-31:
latest mk client for Linux amd64 does not work for me too - I see a screen with progress bar and then only empty grey window. The output:
```
./MachinekitClient_Development-201707270829-master-eb599d1-x64.AppImage             
qml: Starting to load the main application.                                                                                          
qrc:///main.qml:33:1: plugin cannot be loaded for module "Machinekit.HalRemote": Cannot load library /tmp/.mount_nNfK3p/qml/Machinekit/HalRemote/libmachinekithalremoteplugin.so: (/usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /tmp/.mount_nNfK3p/lib/libzmq.so.5))                                                                                                
qml: Finishing the splash screen progress bar.                                                                                       
qml: Hiding the splash screen.                                                                                                       
qml: Deleting the splash screen.
```

machinekoder, 2017-07-31:
Yes, I moved the build servers to Debian Stretch. Seems to be a GCC5 issue on older distros. I will see what I can do.

machinekoder, 2017-08-03:
Just tested on Jessie. Problem should be fixed with https://github.com/qtquickvcp/QtQuickVcp/pull/184

propcoder, 2017-08-03:
Yes, it works on jessie, but interesting that it does not launch by mouse
from gui and does not have an icon like earlier versions do..

closed 2017-08-04 by machinekoder