machinekit/QtQuickVcp #178: Protobuf installation on Linux

propcoder, 2017-06-30:
First issue: curl not found AFAIK can be solved easily by
`sudo apt-get install curl`

~/protobuf$ ./autogen.sh
Google Test not present.  Fetching gtest-1.5.0 from the web...
./autogen.sh: 22: ./autogen.sh: curl: not found

bzip2: Compressed file ends unexpectedly;
        perhaps it is corrupted?  *Possible* reason follows.
bzip2: Inappropriate ioctl for device
        Input file = (stdin), output file = (stdout)

It is possible that the compressed file(s) have become corrupted.
You can use the -tvv option to test integrity of such files.

You can use the `bzip2recover' program to attempt to recover
data from undamaged sections of corrupted files.

tar: Child returned status 2
tar: Error is not recoverable: exiting now
machinekit@kach-jess:~/protobuf$ ./configure
bash: ./configure: No such file or directory

Second issue can be solved by:
`curl -L https://github.com/google/googletest/archive/release-1.5.0.tar.gz | tar zx
mv googletest-release-1.5.0 gtest`
Found on [https://github.com/datacratic/protobuf/issues/2](url)
Google Test not present.  Fetching gtest-1.5.0 from the web...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1586  100  1586    0     0  22676      0 --:--:-- --:--:-- --:--:-- 22985
bzip2: (stdin) is not a bzip2 file.
tar: Child returned status 2
tar: Error is not recoverable: exiting now

machinekoder, 2017-06-30:
If you are using Debian Jessie or newer you can also install the protobuf package from the distro:

```bash
sudo apt install protobuf-dev protobuf-compiler
```

machinekoder, 2017-06-30:
Or a newer version of Protobuf in general, I just uploaded a patch that fixes problems with Protobuf 3.x.

propcoder, 2017-06-30:
On Jessie 4.1.19-rt22mah #1 SMP PREEMPT RT Tue Mar 15 09:12:20 CET 2016 i686 GNU/Linux:
```
sudo apt install protobuf-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package protobuf-dev
```

machinekoder, 2017-07-21:
@propcoder Sorry for the typo, its `libprotobuf-dev`

machinekoder, 2017-07-27:
@propcoder Can I close this issue?

_Sent from my Verizon SM-G930V using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

propcoder, 2017-07-27:
I don't know, because I built from sources and I don't want to mess up the
working system now. How should I check that? libprotobuf-dev is already
installed on my laptop.

On 27 July 2017 at 09:44, Alexander Rössler <notifications@github.com>
wrote:

> @propcoder <https://github.com/propcoder> Can I close this issue?
>
>
>

machinekoder, 2017-07-28:
Then I guess we can close it since the issue was about the Protobuf installation instructions not working.

closed 2017-07-28 by machinekoder

sirop, 2017-07-31:
```
git checkout v3.3.0
./autogen.s
./configure
make 
sudo make install
```
works on Ubuntu 14.04 .