machinekit/QtQuickVcp #180: AND demo with MachinekitClient_Development-201707110852-master-fd6e167-x64  on Windows 8.1

sirop, 2017-07-20:
AND demo does not really load with MachinekitClient_Development-201707110852-master-fd6e167-x64  on Windows 8.1
[image]

Screen capture: https://youtu.be/5-r3bfQ6uas
Although the 8th second shows that two buttons and an LED are loaded, this state does not sustain itself.

The latest Android build works well, so this can not be a problem of my local network settings.

BTW: is mklauncher and machinekit client have always to be used now or is this tutorial
https://github.com/machinekit/machinekit/tree/master/src/machinetalk/tutorial/motorctrl  -- which does not mention either mklauncher or MK client -- still valid?

machinekoder, 2017-07-20:
@sirop If Multicast is not working try the Unicast mode.

The tutorial does not contain the QtQuickVcp UI, so it might be outdated.

sirop, 2017-07-20:
@machinekoder 
That is already Unicast mode. It seems as the 8th second of the screen capture shows that the connection is established for a fraction of a second but then the connection breaks again.

machinekoder, 2017-07-20:
@sirop No, I don't think it was connected, the loading screen is on top of the actual UI so it may be the case that it took some time to load.

It looks like the haltalk services aren't found. Can you please show me the output of the Service Window (local UIs).

Also, make sure you have the latest and greatest version of MachinekitClient, there was a bug that sounds like this in an older version.

It could also be the case that the Windows builds failed - which I have to check.

sirop, 2017-07-21:
You seem to be right:
[image]

I do use the latest build from your build server.

linuxcnc.log if of any help:
```
Jul 21 03:08:04 debian-slave haltalk: register_stuff: actual hostname as announced by avahi='debian-slave.local'
Jul 21 03:08:04 debian-slave msgd:0: zeroconf: registered 'Log service on debian-slave.local pid 2008' _machinekit._tcp 49152 TXT "uuid=72c3f16f-59c3-4116-ba4f-709959abb157" "instance=7be292de-6dfc-11e7-8b6f-c4e9841daedd" "service=log" "dsn=tcp://debian-slave.local:49152"
Jul 21 03:08:04 debian-slave msgd:0: hal_lib:2023:user haltalk: talking HALRcomp on 'dsn=tcp://%s:49154'
Jul 21 03:08:04 debian-slave msgd:0: hal_lib:2023:user haltalk: talking HALComand on 'dsn=tcp://%s:49155'
Jul 21 03:08:04 debian-slave haltalk: zeroconf: registering: 'HAL Group service on debian-slave.local pid 2023'
Jul 21 03:08:04 debian-slave haltalk: zeroconf: registering: 'HAL Rcomp service on debian-slave.local pid 2023'
Jul 21 03:08:04 debian-slave haltalk: zeroconf: registering: 'HAL Rcommand service on debian-slave.local pid 2023'
Jul 21 03:08:05 debian-slave haltalk: zeroconf: registered 'HAL Group service on debian-slave.local pid 2023' _machinekit._tcp 49153 TXT "uuid=72c3f16f-59c3-4116-ba4f-709959abb157" "instance=7c5b6344-6dfc-11e7-8b6f-c4e9841daedd" "service=halgroup" "dsn=tcp://debian-slave.local:49153"
Jul 21 03:08:05 debian-slave haltalk: zeroconf: registered 'HAL Rcomp service on debian-slave.local pid 2023' _machinekit._tcp 49154 TXT "uuid=72c3f16f-59c3-4116-ba4f-709959abb157" "instance=7c5b6344-6dfc-11e7-8b6f-c4e9841daedd" "service=halrcomp" "dsn=tcp://debian-slave.local:49154"
Jul 21 03:08:05 debian-slave haltalk: zeroconf: registered 'HAL Rcommand service on debian-slave.local pid 2023' _machinekit._tcp 49155 TXT "uuid=72c3f16f-59c3-4116-ba4f-709959abb157" "instance=7c5b6344-6dfc-11e7-8b6f-c4e9841daedd" "service=halrcmd" "dsn=tcp://debian-slave.local:49155"
```

I opened 49152-49155 ports in the Windows firewall but that did not help.

machinekoder, 2017-07-21:
Can please you also post the output of `hostname`. There was an issue with UpperCased host names.

sirop, 2017-07-21:
Yes, on Windows it is uppercased:
```
$ hostname
MININT-TGHD2NK
```

machinekoder, 2017-07-21:
How does the casing of the `debian-slave` look like? (The machine which runs Machinekit).

sirop, 2017-07-21:
It is all lower case:
```
$ hostname
debian-slave
```

sirop, 2017-07-27:
Services not discovered was a good hint.
Solution is to set up permissive rule for UDP port 5353  for inbound connection in Windows firewall.

closed 2017-07-27 by sirop