machinekit/QtQuickVcp #182: Buttons strange on new Android device

luminize, 2017-07-24:
Minor Issue:
I've purchased a Samsung Galaxy Tab and have installed Machinekit-Client from the Google Play Store.
I notice that buttons which normally have an uniform gray color, these show as bars with alternating color. Is this something Android specific?

machinekoder, 2017-07-24:
In short - yes, thats the Android Material style.

Bas de Bruijn writes:

> Minor Issue:
> I've purchased a Samsung Galaxy Tab and have installed Machinekit-Client from the Google Play Store.
> I notice that buttons which normally have an uniform gray color, these show as bars with alternating color. Is this something Android specific?

machinekoder, 2017-07-24:
You can of course create your own Buttons if you want or choose an alternative Qt style if you can compile the from source. http://doc.qt.io/qt-5/qtquickcontrols-platformnotes.html#android-style

luminize, 2017-07-24:
Ah, ok, thanks. I was thinking it was something going wrong on my side.

closed 2017-07-24 by luminize