machinekit/QtQuickVcp #218: Android:  "QML Error:  Loading QML file failed"

zultron, 2017-10-31:
I'm getting this error (on Android 6.0.1, Machinekit app v0.9.7.1 from Play Store) trying to launch the [Goldibox][1] configuration.  The same configuration works fine from the Linux x86 AppImage.

The "Available Launchers" "Big" list looks slightly different, because the icon does not show up.  This makes me wonder if the failure comes from the .svg images used both for the icon and for GUI elements, but ultimately this is a wild guess.

How do I debug this?

[1]: https://github.com/zultron/goldibox

zultron, 2017-11-01:
Switched background graphic to png format, but same error message.

machinekoder, 2017-11-01:
I agree the error message created by QML Loader is not very helpful. The only way to debug this at the moment is to build and run the Android version to see the console output of the application.

The version in the Play store was one commit behind the master, maybe that makes a difference as well.

machinekoder, 2017-11-01:
Alternatively starting the machinekit.io.appdiscover application with adb from the command line might also show more interesting messages.

machinekoder, 2017-11-02:
@zultron This patch here will help you: https://github.com/qtquickvcp/QtQuickVcp/pull/221
Needs some time to propagate to Android though.

machinekoder, 2017-11-03:
@zultron Android version is being rolled out right now. You should see the real error very soon.

zultron, 2017-11-03:
@machinekoder Thanks so much!  Here's the full error:

```
QML Error:
Loading QML file failed:
file://data/user/0/io.machinekit.appdiscover/files/Machinekitclient-ep9vs0/goldibox-remote.qml:78 Type HalDial unavailable
file:///data/data/io.machinekit.appdiscover/qt-reserved-files/qml/Machinekit/HalRemote/Controls/HalDial.qml:51 Type Dial unavailable
file:///data/data/io.machinekit.appdiscover/qt-reserved-files/qml/Machinekit/Controls/Dial.qml:266 Styleitem is not a type
```
Now, if that's referring to [Dial.qml][1], that file hasn't changed for a year, does it mean I did something wrong?

[1]: https://github.com/qtquickvcp/QtQuickVcp/blob/master/src/controls/Dial.qml#L266

machinekoder, 2017-11-04:
I have added the Dial Control a long time ago from Qt's private QtQuickItem. Unfortunately, this control is not available on Android -  and should possibly be deprecated in general.

I recommend you to switch to [HalKnob](https://github.com/qtquickvcp/QtQuickVcp/blob/master/src/halremotecontrols/HalKnob.qml) instead.

QtQuick.Controls 2.0 also have Dial control that is cross-platform compatible: <https://doc.qt.io/qt-5/qml-qtquick-controls2-dial.html>. In combination with a HalPin it should do what you need. See the implementation of HalRemote.Controls for reference how to create your custom HAL enabled control.

zultron, 2017-11-05:
Thank you @machinekoder.  The `HalKnob` did the trick, and the useful error messages helped me debug a different problem as well.

closed 2017-11-05 by zultron