machinekit/QtQuickVcp #219: gcode program preview doesn't move after touch off

machinekoder, 2017-11-01:
Moved froml: https://github.com/qtquickvcp/Cetus/issues/7

After loading any ngc file, live preview is probably set to 0,0 machine coordinates. And doesn't move when doing touch off on any axis (coordinate origin is moved tho). While doing same operation in axis GUI coordinate origin is moved to current tool location and preview is moved related to touched off position. Is this intended or a bug in preview?Client tested on all OS'es: OSX sierra, windows 10 64bit, ubuntu 16.04 64bitmachinekit running on debian jessie
--

machinekoder, 2017-11-01:
@spaivaras This behavior is not intended. Instead, the preview should be refreshed after a touch of operation or the preview of offset updated - whatever is more applicable.

spaivaras, 2017-11-01:
I am looking into this myself but my qt knowledge is very basic. It seems that after touch off no updates are called to GLPathItem::calculateNewPosition which seems to be handling g5x offsets. If I will find anything usefull will post here

PR #220 (2017-11-01): Change programExtents and path position to g5x offset
Not sure if is actually a valid solution, but it works for me :) this should fix #219. Not sure how to to run tests, maybe this PR could be an inspiration for someone with actual knowledge here ;)
closed 2017-11-01 by machinekoder