machinekit/QtQuickVcp #248: Non-existent paths shown in Cetus preview window.

JTrantow, 2017-12-01:
I generate g-code with Fusion 360 using linuxcnc.cps - LinuxCNC (EMC2). I can run axis on the BBB (slowly and with a lot of restrictions) but it displays everything as expected. 

When I use the Windows Cetus client with some g-code, I see extra paths (vertical curves) in the preview. The g-code runs correctly, it's just a preview issue. 

Here is a subset of the g-code that illustrates the problem.
%
(1001)
(T1  D=6.35 CR=0. - ZMIN=-20.05 - FLAT END MILL)
N10 G90 G94 G17 G91.1
N15 G21
N20 G53 G0 Z0.

N2455 G0 Z15.
(CIRCULAR1)
N2465 M8
N2470 G0 X58.737 Y-0.635
N2475 Z15.
N2480 Z-3.
N2485 G1 Z-4.365 F1524.
N2490 G18 G2 X59.373 Z-5. I0.635 K0.
N2495 G1 X59.69
N2500 G17 G3 X60.325 Y0. I0. J0.635
N2505 X-60.325 I-60.325 J0.
N2510 X60.325 I60.325 J0.
N2515 X59.69 Y0.635 I-0.635 J0.
N2520 G1 X59.373
N2525 G18 G3 X58.737 Z-4.365 I0. K0.635

N9000 M30
Inline image 1
Inline image 1

machinekoder, 2017-12-01:
I can confirm the problem
[image]
G2 preview needs some fixes.

machinekoder, 2017-12-02:
Okay, so G2/G3 preview for planes other than XY is broken for now. I tried to fix it, but this needs a lot more though. Part of the problem is inside Machinekit, fixes are underway. The ARC_FEED preview commands are somewhat confusing/

JTrantow, 2017-12-02:
I'm not sure if it is related, but I also noticed the extents did not
include the extents of the arc. I will check if axis does this correctly.

machinekoder, 2018-02-18:
@JTrantow A fix is upcoming: https://github.com/machinekit/QtQuickVcp/issues/248

JTrantow, 2018-02-20:
I updated the 64 bit client on my Windows 10 desktop, and Cetus and
MachineFace on the BBB.

The Client sees the interfaces but when I open Cetus or MachineFace, it
just hangs Waiting for services to appear.  I went back through all my
settings and couldn't find any problems. I have just been changing the
DISPLAY = axis to mkwrapper to go back and forth.
[image: Inline image 1]

As I was preparing this email, I tried the Android MachineKit App (which I
have not updated in a while) and it still starts up Machineface and Cetus
as expected. I just have a small Android Pixel screen so it's difficult to
see much on the Cetus preview screen. But it doesn't hang waiting for
services like the Windows Client.

I installed the Windows client on a virgin Window 10 laptop and it also
hangs "Waiting for services to appear..."
I closed the Windows client and opened up the Android Machine kit client
and it works fine.

Looks like something is wrong with the Windows client? I was running the
latest 64bit version on both Windows 10 computers.

I can run the And2demo

JTrantow, 2018-02-20:
I just installed the latest Android client and the Cetus and Machineface
interfaces still work.

JTrantow, 2018-02-23:
Window32 client does not work. Stuck waiting for services just like 64 bit.

I use Fusion360 with the LinuxCNC post processor and set the WCS working
coordinate system to be at the center of the top radius.
[image: Inline image 2]

My experience with LinuxCNC Axis is that when I load g-code the WCS will be
at the zero current coordinates. I normally use G54 touch off and sometimes
modify that with G92.
I configured a LinuxCNC box with a similar machine configuration and
everything works as expected.

MachineKit has several problems relative to LinuxCNC that show up with
g-code generated by Fusion 360 (using the LinuxCNC post processor)
Something in the .ngc causes
1.) the origin of the loaded .ngc to load with an offset (The G54, G92
offsets and tool show correctly, its just the loaded file that is offset.)
2.) Extraneous paths that are not in the g-code show up and throw off the
path dimensions.
3.) When g-code runs the machine moves to the correct coordinates but
sometimes the code won't run if the offset paths are outside the machine
limts.

I loaded several of the example .ngc files and they don't have these
problem. All the files that exhibited the problem with MachineKit work fine
with LinuxCNC.

The Windows Client is no longer working for me and the Android client on my
phone is too small to see details well, so I created a VMWare Ubuntu
17.10.1 VM and used the Linux 44 Client.

I am pretty sure the offsets are incorrect (and extra paths show up with
the attached file.)
I base this on loading the gcode and seeing where the gcode coordinates end
up using the preview. The machine coordinates and the relative offsets
always look correct and I have cut gcode when the preview shows the g-code
with the offset. (tool and machine travel is correct)
In the diagram below, The machine coordinates are correct and the tool is
displayed correctly. But when I load the G-code, I expect the origin at the
center of the Maybe Machinekit is simply misinterpretting the origin in the
Fusion file?

[image: Inline image 1]

Fusion360 has a bunch of options for postprocessing the g-code. This might
be useful for temporarily disabling whatever g-code feature is confusing
Machinekit. Axis seems to have same or similar problems to the remote UIs.

[image: Inline image 2]
[image: Inline image 3]
I will leave my BBB configuration setup intact and will test things out for
you but I need to shift my focus to cutting some prototypes with linuxcnc.
I will give up on the BBB for the near future as the microATX with LinuxCNC
is faster and more stable than what I have managed with the BBB. The other
big problem I had with the remote clients was I lost the local pendant
control when using the remove client.

machinekoder, 2018-03-03:
The patch is the current MachinekitClient now, this also requires a machinekit update.

closed 2019-11-21 by JTrantow