machinekit/QtQuickVcp #249: Tool in Preview is not using same coordinates as display g-code paths

JTrantow, 2017-12-01:
I normally use a G92 to set the origin somewhere within my router table. After a G1 X0, Y0 the following should show the tool at the origin of the loaded g-code.  When the code executes I see the tool moving offset from the g-code circles.
So it appears the tool display isn't using the same coordinates(G92) as the g-code in the preview. 

Inline image 2

machinekoder, 2017-12-01:
g92 was indeed not used in the preview: https://github.com/qtquickvcp/QtQuickVcp/pull/250

machinekoder, 2017-12-01:
The issue should be fixed with the next build.

machinekoder, 2018-01-01:
@JTrantow Can you please verify if the problem has been resolved.

JTrantow, 2018-01-02:
Are the changes just in qt or deeper in machinekit? Just let me know what
repos I should update and I will verify in the next day or two.

machinekoder, 2018-01-03:
@JTrantow The changes are in QtQuickVcp only,

JTrantow, 2018-01-05:
I took a quick look tonight and I'm confused with updating QtQuickVcp.

A few weeks ago I gave up on the BBB rt-preempt and reverted to the
machinekit Jessie 3.8 xenomai image (to get webcam working). I have cloned
QtQuickVcp/Machineface and QtQuickVcp/Cetus. I understand how to update
those clients, but I'm not sure about updating QtQuickVcp itself??  I
thought that was a machinekit component and I don't recall installing
QtQuick by itself??

I did a sudo apt-get update and
The following packages have been kept back:
  libdrm-amdgpu1 libdrm-dev libdrm-etnaviv1 libdrm-exynos1
libdrm-freedreno1 libdrm-nouveau2 libdrm-omap1 libdrm-radeon1 libdrm-tegra0
libdrm2 libgl1-mesa-dri libkms1 machinekit
  machinekit-xenomai mesa-va-drivers

I tried sudo apt-get install machinekit-xenomai and sudo apt-get install
machinekit

I didn't see anything obvious about QtQuickVcp gettting updated and the
problem still exists so I don't think I have the update in place.

I don't understand the Rolling Releases section at
https://github.com/qtquickvcp/QtQuickVcp. Do I need to install a QtQucikVcp
library on the BBB? How? (I'm not that familiar with Linux.)

Sorry for all the newbie questions. I actually set up linuxCNC on an old PC
because of the offset and because my pendant doesn't work when I use Cetus.
But I switched back to BBB to test this out.

machinekoder, 2018-01-05:
QtQuickVcp is independent of the mainstream Machinekit packages and has it's own release cycle.

At the moment there is no QtQuickVcp package for the BBB as the graphics driver just don't work very well.

However, QtQuickVcp is designed as remote UI. So just download the rolling release MachinekitClient from the [download section](https://github.com/qtquickvcp/QtQuickVcp#rolling-releases) to your computer and connect remotely.

There is no need to install or update anything QtQuickVcp related on the BBB itself (except the UIs which are remote deployed).

You might also wan't to check out Machinekit on Android in case you have an Android tablet.

JTrantow, 2018-01-05:
> In that case, I don't think the fix completely worked with the Window
> Cetus???
>
> I updated the MachinekitClient on my Windows machine.
>
> [image: Inline image 2]
>
> [image: Inline image 1]
>
>
> The tool displays in the correct location, but the g-code paths still show
> up with an offset. When I start cutting, the tool moves to the correct
> coordinates, but the g-code display is offset and the dimensions are bigger
> than expected. The machine envelope is correct and the tool is correct.
> [image: Inline image 3]
>
>
> I can run the Cetus UI on my Android phone. I updated with the Android
> Play Store. It's hard to see the preview on my phone screen, but the tool
> and paths do line up in the preview which is different than the Windows
> Cetus?
> The paths show the extra arc and lines that I think are throwing off the
> dimensions of the cutting paths.
> [image: Inline image 4]
>
> With the same .ngc file, Axis shows the paths and tool correctly. It also
> shows the y dimensions correctly (the arc erroneously displayed by the
> remote UIs doesn't show up in Axis) but the x dimension is larger than it
> should be and it has the extraneous horizontal line extending to the right
> of the paths.
>
> On Fri, Jan 5, 2018 at 1:08 PM, Alexander Rössler <

closed 2019-11-21 by JTrantow