machinekit/QtQuickVcp #256: Build error in ConnectionWindow.qml

zultron, 2018-01-25:
With Qt v. 5.7.1 from Debian Stretch packages, and installing QQVCP from source, running an app from qtcreator ("Debug" mode) fails with the following errors:
```
QQmlApplicationEngine failed to load component
qrc:///main.qml:15 Type ConnectionWindow unavailable
file:///usr/lib/x86_64-linux-gnu/qt5/qml/Machinekit/Application/Controls/ConnectionWindow.qml:183 Unexpected token `<'
file:///usr/lib/x86_64-linux-gnu/qt5/qml/Machinekit/Application/Controls/ConnectionWindow.qml:183 Expected token `:'
```

The line in question is here:
    https://github.com/machinekit/QtQuickVcp/blob/master/src/applicationcontrols/ConnectionWindow.qml#L183

This looks like good syntax, judging from the Qt docs.  Is this error the result of the older Qt version packaged in Stretch?

zultron, 2018-01-25:
Hmm, doesn't seem to be the version; the [v. 5.6 docs][1] show the following legal syntax:

```
[default] property list<<objectType>> propertyName
```

[1]: http://doc.qt.io/qt-5.6/qtqml-syntax-objectattributes.html#property-attributes

zultron, 2018-01-25:
If the `default` keyword is dropped, the run succeeds:

`property list<ApplicationDescription> applications`

At this point I'm out of my depth.  Does this make any sense?  Thanks!

sirop, 2019-02-05:
The same on RPI3 with Qt 5.7.1. I start the compiled machinekit-client:
```

libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
JIT is disabled for QML. Property bindings and animations will be very slow. Visit https://wiki.qt.io/V4 to learn about possible solutions for your platform.
qml: Starting to load the main application.
qrc:///main.qml:51:5: Type ConnectionWindow unavailable
file:///home/pi/build_qtquickvcp/QtQuickVcp/imports/Machinekit/Application/Controls/ConnectionWindow.qml:183:26: Unexpected token `<'
file:///home/pi/build_qtquickvcp/QtQuickVcp/imports/Machinekit/Application/Controls/ConnectionWindow.qml:183:49: Expected token `:'
```

zultron, 2019-02-07:
I believe this is due to the syntax being introduced only in later Qt versions.  I dabbled with fixing the code in QtQuickVcp and with building newer Qt5 from source, but ultimately didn't hit upon an elegant solution, and ended up switching over to the MachinekitClient bintray images on `amd64`.

Since this one-liner is apparently the only thing keeping QtQuickVcp from building against Debian Stretch packages, I wonder if fixing it would lower the bar for folks to start contributing.

ArcEye, 2019-02-07:
QML leaves me cold, but looking at the docs you linked to, it seems a list should have double \<\< around it and the declaration of a property  just has single \<.

There is also an inconsistency of white space, where double arrowheads have no spacing between the type and the first arrowhead and single arrowheads are spaced from the preceding type, but that could just be style on the part of the example writer.

machinekoder, 2019-02-07:
Yes, the syntax is from newer Qt versions. Just replace it with `var` for a quick fix.

machinekoder, 2019-02-15:
Update changing things to `var` breaks the default property. In case you set it to `var`, make sure also to change the uses of `ConnectionWindow` to use `applications:` instead of the `default property.`