machinekit/QtQuickVcp #278: cstdlib:75:15: fatal error: stdlib.h: No such file or directory

l29ah, 2018-08-11:
Trying to build QtQuickVcp using gcc-7.3.0, getting a bunch of such errors:
```
x86_64-pc-linux-gnu-g++ -c -O2 -pipe -march=nehalem -mtune=ivybridge -w -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -I. -Ibuild/cpp -isystem /usr/include -I/usr/lib64/qt5/mkspecs/linux-g++ -o emcclass.pb.o build/cpp/machinetalk/protobuf/emcclass.pb.cc
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/include/g++-v7/ext/string_conversions.h:41:0,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/include/g++-v7/bits/basic_string.h:6349,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/include/g++-v7/string:52,
                 from build/cpp/machinetalk/protobuf/types.pb.h:7,
                 from build/cpp/machinetalk/protobuf/types.pb.cc:4:
/usr/lib/gcc/x86_64-pc-linux-gnu/7.3.0/include/g++-v7/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
               ^~~~~~~~~~
```
Removing `-isystem /usr/include` from the compiler args fixes it.

ArcEye, 2018-08-12:
This occurs throughout Qt in later versions with newer compilers.

The way to fix it is to edit mkspecs/common/gcc-base.conf

Change `QMAKE_CFLAGS_ISYSTEM = -isystem`
to `QMAKE_CFLAGS_ISYSTEM = -I` (capital i)

I have been doing this since Qt 5.10.0 at least

l29ah, 2018-08-12:
Thanks! I found out it's simpler to just call `qmake 'QMAKE_CFLAGS_ISYSTEM=-I'`.

ArcEye, 2018-08-12:
If I don't fix it at source, I am liable to waste time the next time it occurs, remembering what I did to fix it last :stuck_out_tongue_closed_eyes:

insangi, 2019-08-02:
@ArcEye and @l29ah 
This post helped me get rid of stdlib.h issue.
Do you mind helping me I am trying to make QT for tx1 using GCC. I am getting this error at the end. Please help me solve this. Thank you

/Downloads/gcc-linaro-7.4.1-2019.02-x86_64_aarch64-linux-gnu/bin/../lib/gcc/aarch64-linux-gnu/7.4.1/../../../../aarch64-linux-gnu/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
Makefile:1179: recipe for target '../../lib/libQt5Core.so.5.14.0' failed
make[2]: *** [../../lib/libQt5Core.so.5.14.0] Error 1
make[2]: Leaving directory '/home/ibrahim/Downloads/qtbase/src/corelib'
Makefile:198: recipe for target 'sub-corelib-make_first' failed
make[1]: *** [sub-corelib-make_first] Error 2
make[1]: Leaving directory '/home/ibrahim/Downloads/qtbase/src'
Makefile:50: recipe for target 'sub-src-make_first' failed
make: *** [sub-src-make_first] Error 2

ekubay, 2020-02-15:
hi guys where can I find this "QMAKE_CFLAGS_ISYSTEM = -isystem"  
to QMAKE_CFLAGS_ISYSTEM = -I (capital i)
 please help me

surajmahangade, 2020-06-13:
Run
`sudo find / -name "gcc-base.conf"`
as @ArcEye  mentioned change file in this location mkspecs/common/

vishnuu95, 2020-07-16:
I had the same problem recently. I believe the issue was due to multiple gcc versions. A easy fix was to navigate to c++/8/ folder and replace all #include_next with #include . The files stdlib.h are present in the same directory itself and hence #include does the job.