machinekit/QtQuickVcp #285: Hal Remote Component Usage

pauloasherring, 2019-03-28:
Hey there! 
First of all, I have to start congratulating you for this wonderful piece of software! It sure is a usage case for me!
Regarding my issue: I am trying working on a custom GUI, based on QtQuickVCP. I am implementing a manual tool change routine, based on remote hal components. I am using Cetus as a starting point. The following code is supposed to access the remote components:
`            RowLayout {
                id: container
                anchors.margins: 5
                Label{
                    text: "testingLabel"
                }

                HalButton{
                    name: "userChangedTool"
                    Layout.alignment: Layout.Center
                    text: qsTr("Continue")
                    checkable: false
                    //enabled: toolChangeButtonEnable.value

                }
                HalLed {
                    id: toolChangeButtonEnable
                    Layout.alignment: Layout.Center
                    name: "toolChangeButtonEnable"
                }
                Service {
                    id: halrcompService
                    type: "halrcomp"
                }
                Service {
                    id: halrcmdService
                    type: "halrcmd"
                }
                HalRemoteComponent {
                     halrcmdUri: halrcmdService.uri
                     halrcompUri: halrcompService.uri
                     ready: (halrcmdService.ready && halrcompService.ready) || (state === HalRemoteComponent.Connected)
                     name: "toolChangeControl"
                     containerItem: container
                }
            }`
The issue is that if I add the code to a existing tab, such as inside the "ManualTab.qml", it does not work and if I add it to its own tab, is works without a glitch. Given that it works normally in its own tab, I think that it might have something to do with my implementation itself. It might be interacting along with other elements added to the tab, such as "ApplicationObject" or "ApplicationCore".
in that case, what should be the approach to get a remote component and a applicationcore component to work together?
Best Regards.
Paulo Sherring.

alkabal, 2019-05-06:
Hi

I can’t help you but I like to test manual toolchange if tou are ok.

https://github.com/machinekit/Cetus/issues/23

Br

machinekoder, 2019-10-12:
Make sure to create the contaierItem "container" reference by the HalRemoteComponent and that this Item is not contained in another containerItem/HalRemoteComponent.

Inside a Tab, also make sure that the Tab is set to `active: true`. Best take a look at the Cetus and Machineface source code for some examples of multiple HAL remote components.