machinekit/QtQuickVcp #291: Machinekit Client Android connect to existing session

dieterdanger, 2019-09-30:
Hello everyone,
we're using machinekit 0.1.1561909734.gitc4fb096-1~stretch on Debian Stretch. 

Right now, we have a dedicated 3D-Printer PC right next to the printing machine where we open the Machinekit Client for Linux appimage. This works flawlessly. It is also possible to connect remotely from another linux machine to the same session. All inputs that are made on one interface are perfectly mirrored on the other.

But, if we are trying to connect using the android app (unicast), it fails as long as any Machinekit Client is running. Any idea why this is happening? It would be good if we could observe printing progress via android app.

Best regards
dieterdanger

sirop, 2019-09-30:
Just out of curiosity: does multicast on Android work with your setup?

machinekoder, 2019-10-12:
Interesting, I've heard other users reporting the same problem, but I couldn't reproduce it.

@dieterdanger What do you mean with VPN problem?

dieterdanger, 2019-10-14:
Our machinekit pc is connected to the 3d printer in a specific subnet via
lan cable. Since our work policies do not allow WLAN access into this same
subnet, a connection to this subnet has to be established via VPN. What
follows now are assumptions :-) Since Multicast relies on ?broadcast?, I
suspected VPN to be the issue, because other things like wake on lan are
also not working. But like I said, I do not have a deeper understanding in
this matter.

For us, it would be sufficient if Unicast could overtake an established
session, which right now doesn't seem to be possible.

Thank you for all your great work Alexander!

Best regards