machinekit/QtQuickVcp #301: How to get QtQuickVcp

summerlotus513, 2021-12-13:
Hello
I am trying to use QtQuickVcp, but I encountered some problems when compiling
The binary file download was once provided in the Readme, but https://bintray.com/ seems to have been migrated to https://jfrog.com, and I cannot get the compiled QtQuickVcp for each platform. May I ask where should I be? Download them?

sirop, 2021-12-13:
Fork this repo. Merge https://github.com/machinekit/QtQuickVcp/pull/299/files into your fork.

sirop, 2021-12-13:
Se  also https://github.com/gb-g/QtQuickVcp/releases.

summerlotus513, 2021-12-14:
> 

marvelous!
I'm trying out msvc2017 to build related content. Can the files provided by you support it?
I copied the files in the compressed package to the specified directory (such as qml/Machinekit -> (MyQtInstallDir)/(Version)/msvc2017/qml), and then opened a project, but he still prompted
`
QML module does not contain information about components contained in plugins.
Module path: D:/Qt/Qt5.12.9/5.12.9/msvc2017/qml/Machinekit/Application/Controls
See "Using QML Modules with Plugins" in the documentation.
Automatic type dump of ML module failed.
First 10 lines or errors:
"D:\Qt\Qt5.12.9\5.12.9\msvc2017\bin\qmlplugindump.exe"returned exit code 3.
Arguments:-nonrelocatable Machinekit.Application.Controls 1.0.
QQmlComponent:Component is not ready
file:///C:/Users/zhou/typelist.gml:3:1:module "Machinekit.Application.Controls" is not installed
Check'General Messages' output pane for details.
`
So how should I solve this problem?

sirop, 2021-12-14:
The files provided by @gb-g  seem to be built with msvc 2015.
https://github.com/machinekit/QtQuickVcp/blob/fb60c2bdaf27888ff32dc1f6b3a57785442891a1/appveyor.yml#L24

summerlotus513, 2021-12-15:
> 

ok，I get it.
by the way, Where can I download the library for Android armv7? @gb-g does not seem to provide a download for Android armv7 .

sirop, 2021-12-15:
Theny you have to build the android libs yourself:
https://github.com/machinekit/QtQuickVcp#android_install

summerlotus513, 2021-12-16:
> 然后你必须自己构建 android 库：[https](https://github.com/machinekit/QtQuickVcp#android_install) : [//github.com/machinekit/QtQuickVcp#android_install](https://github.com/machinekit/QtQuickVcp#android_install)

ok
thanks.

closed 2021-12-16 by summerlotus513